PX4/PX4-Autopilot #17808: error in HIL Standard VTOL QuadPlane

mengchaoheng, 2021-06-26:
**Describe the bug**
dispaly "BARO #0 faild : TIMEOUT!" in QGC.

**To Reproduce**
Steps to reproduce the behavior:
just like: https://docs.px4.io/master/en/simulation/hitl.html

[image]

[image]

mengchaoheng, 2021-06-26:
HIL Quadcopter X does not have this problem

mrpollo, 2021-06-30:
@mengchaoheng Which hardware were you testing with?

mengchaoheng, 2021-06-30:
pixhawk 4 and pix 4 mini 

| |
蒙超恒
|
|
***@***.***
|

签名由 网易邮箱大师 定制

On 06/30/2021 23:11, Ramon Roche wrote:

@mengchaoheng Which hardware were you testing with?

mengchaoheng, 2021-07-01:
> @mengchaoheng Which hardware were you testing with?

pixhawk 4 and pix 4 mini

Zhenyu1Wu, 2025-03-16:
Hello, I'm encountering the same issue when conducting VTOL simulation in HITL mode using the Holybro Durandal flight control board with PX4 firmware version 1.13.3. The ground station and MAVROS node keep reporting the error "BARO #0 failed: TIMEOUT!" repeatedly. Have you found a solution to this problem?