PX4/PX4-Autopilot #17811: Gazebo Missing plugin path with custom world

daxaxelrod, 2021-06-28:
**Describe the bug**
We're using custom .world files for simulation testing. To make it more realistic, we're using the Google maps static photo api trying to overlay real world data using the example [here](http://gazebosim.org/tutorials?tut=static_map_plugin&cat=build_world).

We are able to run gazebo our_world_name.world and the plane loads as expected. Also noticed that the map shows up under gazebo's "models" folder in the sidebar. When we run ```make px4_sitl gazebo___our_world_name``` we just get the drone freefalling through space. When we run through px4 we also dont see the map in our plugins folder

We noticed that GAZEBO_* env variables are missing on the standard make calls. After going to /usr/share/gazebo-11 and running ```source setup.sh```, the plugin paths are then filled out and we see both the drone and plane as normal

**To Reproduce**
Steps to reproduce the behavior:
1. Create custom .world file, stick it in /Tools/sitl_gazebo/worlds and add the new world to the /platforms/posix/cmake/sitl_target.cmake file (around line 148)
2. run the world alone: gazebo my_custom_world.world
3. run with the make command:  make px4_sitl gazebo___ my_custom_world
4. See drone falling off the edge of the earth when running through make
5. To fix, run the setup script: /usr/share/gazebo-11/setup.sh
6. Now that env variables (most importantly the plugin path) are properly setup, it should work

What I dont understand is that other .world scripts in the standard /Tools/sitl_gazebo/worlds folder contain plugins and they work just fine. Left me scratching my head for the longest time before I found the setup.sh

Jaeyoung-Lim, 2021-07-07:
@daxaxelrod The setup script run by the make files are https://github.com/PX4/PX4-Autopilot/blob/master/Tools/setup_gazebo.bash and it is being invoked in https://github.com/PX4/PX4-Autopilot/blob/fef2c4339577c1bccd33c649d63a4a31c8005ef7/Tools/sitl_run.sh#L112

It would be great if you can maybe contribute the world with the goolge maps static photo api in https://github.com/PX4/PX4-SITL_gazebo so we can officially support your use case.

Thanks!

daxaxelrod, 2021-07-09:
Hrm interesting. I'll make some edits to the setup script! unfortunate it would be untenable to have a default world that uses the maps api because it requires a private (billable) maps api token.

But the next best thing would to add a section to the docs that tells other people how they can do it themselves!

closed 2021-11-14 by Jaeyoung-Lim