PX4/PX4-Autopilot #17831: Pixhawk Cube not booting (playing "Error Tune") after configuring Iridium Satlink

bernhardpg, 2021-07-01:
**Issue**

After configuring the Pixhawk to use the Iridium module, the Pixhawk refuses to boot if the Iridium module is disconnected. When connected to power over USB, the Cube plays and “Error tune”, indicating a hard fault in the system. See the [PX4 sounds](https://docs.px4.io/v1.9.0/en/getting_started/tunes.html).

**Hardware**

The error was introduced on a Pixhawk Cube Orange, running PX4 v1.11.2, and an Iridium RockBLOCK 9603 module. I have checked, and the problem persists in PX4 v1.12.0-beta6 too, see below.

**Steps to reproduce the error**

1. Make sure the cube is running PX4 v1.11.2, and that a SD card is installed.
2. Power on the Cube using USB
3. Set “ISBD_CONFIG = GPS 2” without connecting the Iridium Module.
4. Reboot the pixhawk
5. It should not refuse to connect to QGC, and will play the “Error Tune“.
6. There should now be a .log file on the SD card which displays a message similar to:[hardfault_log] -- 2000-01-01-00:00:06 Begin Fault Log --System fault Occurred on: 2000-01-01-00:00:06Type:Hard Fault in file:armv7-m/up_hardfault.c at line: 148 running task: iridiumsbd

**Removing the error after introducing it**

1. Disconnect the USB
2. Connect the Iridium module to serial port “GPS 2”
3. It now boots as normal, with the iridium connected.
4. Change “ISBD_CONFIG = disabled”, disconnect iridium, and reboot the vehicle.
5. It now boots as normal, without the iridium connected.

**PX4 Version**
Initially, this error was found on v1.11.2. I tested with the most recent beta of v1.12, v1.12.0-beta6, and the problem still persists. However, the pixhawk does not play an error tune on boot, but reboots automatically continuously and refuses to connect to QGC, but this is the only thing that seems to be different from v1.11.2. Otherwise, the same steps for reproducing the error and removing the error works for v1.12.0-beta6.

**Previous issues**
I found [this](https://github.com/PX4/PX4-Autopilot/issues/10092) issue, which seemed like it could be relevant. However, it was closed and marked as fixed. Not sure if this is a new issue or connected to the previous issue, as I have not had time to debug this any further yet.

LorenzMeier, 2021-07-01:
@RomanBapst Sounds like it is time to get your patch into master so everyone can benefit.

bkueng, 2021-07-02:
@bernhardpg https://github.com/PX4/PX4-Autopilot/pull/17837 should give you a starting point. If you're interested, @RomanBapst can provide the details of the remaining issues.

bernhardpg, 2021-07-02:
Alright, thanks @bkueng @LorenzMeier . I will keep you posted if we develop a fix for this. However, we will probably just avoid the issue altogether by keeping the Iridium plugged in all the time. Hardware tests today also seemed to indicate that this is only an issue with the GPS2 port (and not with TELEM1 or TELEM2), but I did not have time to review that myself.

mrpollo, 2021-08-30:
@bernhardpg is this still an issue for you?

bernhardpg, 2021-08-31:
@mrpollo Yes, the issue persists. We have been working around it by manually changing the parameters and rebooting the pixhawk before we connect/disconnect the iridium module. We are still running v1.11.2

julianoes, 2023-02-10:
It looks like the bootloop error has been resolved in https://github.com/PX4/PX4-Autopilot/pull/17837/commits/6d7e0a10755a88d8a2d876a917df439bd7499982

The output I now get, when trying it, is:

```
Starting Iridium (with MAVLink) on /dev/ttyS5
ERROR [iridiumsbd] modem not responding
ERROR [iridiumsbd] failed to instantiate object
ERROR [iridiumsbd] Timed out while waiting for thread to start
ERROR [iridiumsbd] Task start failed (-1)
```

julianoes, 2023-02-14:
As mentioned above, this should be fixed, closing.

closed 2023-02-14 by julianoes