PX4/PX4-Autopilot #17836: Critical navigation failure! Check sensor calibration

JohnSmithDoc, 2021-07-02:
Hi,I am using px4v1.11 version firmware.I found that if I got "Critical navigation failure! Check sensor calibration"emergency warning in 30 seconds after taking off, the following flags became false but no longer become true even if lpos.xy_valid and lpos.v_xy_valid flag is true.

status_flags.condition_global_position_valid = false;
status_flags.condition_local_position_valid = false;
status_flags.condition_local_velocity_valid = false; 

is this a issue?

I scale some flags in the flightplot:

[image]

[image]

bernhardpg, 2021-07-02:
I remember we got the same error message a while back. In our case, it turned out to be electromagnetic interference messing with the compass. Could be worth taking a look at your magnetometer readings!

JohnSmithDoc, 2021-07-03:
Thank you for your help @bernhardpg 

The reason for this error message is clear to me. It's because the visual_odometry data is bad in 30s after taking off, which caused the three flags:**condition_global_position_valid, condition _local_position_valid and condition_local_velocity_valid** to be false:

[image]

I also have GPS on my flight controller.The flight controller can still rely on GPS data after vision data stopped.Which is clear in the picture below.The EKF fusion is good after using GPS data. The flags : **vehicle_local_position_0.xy_valid and vehicle_local_position_0.v_xy_valid** are true all the time.

[image]

My question is not about the reason of getting that error message.My question is that why the three flags :**condition_global_position_valid, condition _local_position_valid and condition_local_velocity_valid**  did not recover to true after the EKf fusion became good.

Is that a logical bug or is there any logic to this situation?

HHHHHCS, 2021-07-19:
I have the same trouble, this is my maibox

bresch, 2021-07-27:
@JohnSmithDoc It's because when such a failure is detected within the first 30 seconds after takeoff, the local position estimate was considered as faulty. This status cannot be reverted in `v1.11`. See:
 https://github.com/PX4/PX4-Autopilot/blob/21c82a48146fb8afc5a56263a5da21300ba3137e/src/modules/commander/Commander.cpp#L4129-L4130).

With the new yaw emergency estimator and other robustness improvements, we reworked that check to be able to pass again if the EKF recovers.
See: https://github.com/PX4/PX4-Autopilot/pull/16821 and especially 
https://github.com/PX4/PX4-Autopilot/blob/f2ae8ae81466b54cc725c07aa8467991bf74c541/src/modules/commander/Commander.cpp#L3844
This is now in `v1.12`.

JohnSmithDoc, 2021-07-31:
@bresch Got it,thanks!

closed 2021-08-02 by bresch