PX4/PX4-Autopilot #17842: Attitude Estimator doesn't start if internal Compass is disabled

ryanjAA, 2021-07-03:
Tried on Pix 4, Pix 1 and Control Zero on 1.11.3 - all the same.

If you calibrate an external mag, disable the internal one and then unplug the external, on the next reboot (and subsequent) the attitude estimator won't start. 

You need to either plug the external in and reboot or enable the internal. I'll test against 1.12 but I presume it's the same from a similar issue I ran into a month back, just only now able to pinpoint it. 

Not sure if this is intended but at least worth noting as heading not being present makes sense (if there is no compass enabled) but not attitude entirely.

Recreate:

Plug in an external mag
calibrate all sensors
reboot
disable the internal mag
unplug the external mag
reboot

Attitude estimator won't start

ryanjAA, 2021-07-03:
Same on 1.12 beta 6 and master.

ryanjAA, 2021-07-08:
Per conversation with @dagar the ekf2 requires a mag to initialize unless you have EKF2_MAG_TYPE set to none so this isn't a bug, it's just how it was designed so we should just explicitly document it.

@hamishwillee do you have a place that this would make sense? I'm happy to write it (I'll just put a few sentences in so people know what to expect).

What would be great would be QGC would tell the user that no mag is plugged in and/or enabled.

I do see a (separate) issue if you want to fly in stabilized mode with a mag not present though (since ekf2 won't init) but that would be dependent on your board not having a mag or not having a working one (should still be possible though IMO - just lots of warnings since it adds risk if you think you can switch to any heading mode but you can't).

Documentation is the way to go in the short term.

dagar, 2025-11-11:
The attitude does initialize, but GPS position fusion won't start until yaw aligned (mag or GPS heading).