PX4/PX4-Autopilot #17851: VS code issue

zijian98, 2021-07-06:
Hi, I've installed VS code for windows. Opened PX4-Autopilot and installed recommended extensions but VS code always returned this error:  

"Running the contributed command: 'cmake.build' failed."
`[rollbar] Unhandled exception: Unhandled Promise rejection: build Error: Build failed: Unable to configure the project {}`

Any ways to fix this? 
Thanks

dagar, 2021-07-10:
The actual build won't work natively in Win32. Can you try using VSCode remote to build within WSL, Docker, or remotely on a Linux? 

https://code.visualstudio.com/docs/remote/remote-overview

SEUZTh, 2021-09-04:
> The actual build won't work natively in Win32. Can you try using VSCode remote to build within WSL, Docker, or remotely on a Linux?
> 
> https://code.visualstudio.com/docs/remote/remote-overview

Hi, I met the same problem on MacBook M1. I am using Ubuntu20.04 arm64 in Parallels Desktop 17.0.

closed 2025-11-30 by github-actions[bot]