PX4/PX4-Autopilot #17852: Support for NCP5623B - Holybro

vincentpoont2, 2021-07-06:
The current i2c LED driver we using is NCP5623C, but there is a shortage and we are facing sourcing issue. 

NCP5623**B** is available but doesn't look like it's supported, can PX4 add support for NCP5623**B**? 

[NCP5623B Datasheet](https://github.com/PX4/PX4-Autopilot/files/6771546/NCP5623B.pdf)
[NCP5623C Datasheet](https://github.com/PX4/PX4-Autopilot/files/6771545/NCP5623C.pdf)

davids5, 2021-07-08:
I expect it to work. Please test it as follows:
Start the driver with -a 56 (0x38) for the B part and -a 57 (the default 0x39)for the C part.

rgbled_ncp5623c start -X -a 56

```
rgbled_ncp5623c start -X -a 0x38
Usage: rgbled <command> [arguments...]
 Commands:

   start
     [-I]        Internal I2C bus(es)
     [-X]        External I2C bus(es)
     [-b <val>]  board-specific bus (default=all) (external SPI: n-th bus (default=1))
     [-f <val>]  bus frequency in kHz
     [-q]        quiet startup (no message if no device found)
     [-a <val>]  I2C address
                 default: 57
```

closed 2021-07-08 by davids5

vincentpoont2, 2021-07-12:
Hi @davids5 
After adding the line of rgbled_ncp5623c start -X -a 56 to the rc.board_defaults file, compile the firmware, and after flashing into the flight control, the NCP5623B can be recognized.

Can this support be added directly in master and next release?