PX4/PX4-Autopilot #17858: External IMU adis16448 not starting on fmuv3

Jaeyoung-Lim, 2021-07-07:
**Describe the bug**
When trying to run the `adis16448` driver on a fmuv3 hardware, the driver is not able to start.

**Expected behavior**
Driver should start as expected

**Log Files and Screenshots**
Log from boot
```
bad version detected, forcing to fmu-v2
FMUv2 ver 0x8 : Rev 0 V2
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V3
HW type: V2
HW version: 0x00090008
HW revision: 0x00000000
FW git-hash: 0f4eaec6c12dad9a41cea0cb8456f90e88908320
FW version: 1.12.0 0 (17563648)
FW git-branch: develop
OS: NuttX
OS versi[hardfault_log] Fault Log is Armed
on: Release 10.0.0 (167772415)
OS git-hash: 011fbb2ce150e05532e04b2f63d93ee932b226fb
Build datetime: Jul  1 2021 08:14:41
Build uri: localhost
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000100000000393839343335511400460027
MCU: STM32F42x, rev. 3
nsh: cat: open failed: Is a directory
INFO  [param] selected parameter default file /fs/mtd_params
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
INFO  [px4io] IO FW CRC match
Board sensors: /etc/init.d/rc.board_sensors
rgbled #0 on I2C bus 2
WARN  [SPI_I2C] rgbled_ncp5623c: no instance started (no device on bus?)
WARN  [SPI_I2C] hmc5883: no instance started (no device on bus?)
WARN  [SPI_I2C] ms5611: no instance started (no device on bus?)
ms5611 #0 on SPI bus 1 (devid=0x3d)
mpu6000 #0 on SPI bus 1 (devid=0x21)
WARN  [SPI_I2C] mpu9250: no instance started (no device on bus?)
l3gd20 #0 on SPI bus 1 (devid=0x22)
lsm303d #0 on SPI bus 1 (devid=0x11)
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS3
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [px4io] default PWM output device
INFO  [init] Mixer: /etc/mixers/AAERTWF.main.mix on /dev/pwm_output0
INFO  [pwm_out] instance: 0, MAX RATE: 100, default: 50, alt: 50
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1
ERROR [px4io] failed setting PWM rate on IO
ERROR [pwm] PWM_SERVO_SET_SELECT_UPDATE_RATE
ekf2 [476:100]
INFO  [logger] logger started (mode=all)
INFO  [logger] logging 49 topics from logger_topics.txt

NuttShell (NSH) NuttX-10.0.0
nsh> ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 18 -> 25
INFO  [airspeed_selector] No airspeed sensor detected. Switch to non-airspeed mode.

```

**Additional context**
Add any other context about the problem here.

Jaeyoung-Lim, 2021-07-07:
@dagar This is with the debug build enabled (log seems to be identical)

```
reset done, 10 ms

bad version detected, forcing to fmu-v2
FMUv2 ver 0x8 : Rev 0 V2
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V3
HW type: V2
HW version: 0x00090008
HW revision: 0x00000000
FW git-hash: 01307a43feadb97194989961a655d242fe69e77e
FW version: 1.12.0 0 (17563648)
FW git-branch: develop
OS: NuttX
OS versi[hardfault_log] Fault Log is Armed
on: Release 10.0.0 (167772415)
OS git-hash: 011fbb2ce150e05532e04b2f63d93ee932b226fb
Build datetime: Jul  1 2021 08:14:41
Build uri: localhost
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000100000000393839343335511400460027
MCU: STM32F42x, rev. 3
nsh: cat: open failed: Is a directory
INFO  [param] selected parameter default file /fs/mtd_params
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
INFO  [px4io] IO FW CRC match
Board sensors: /etc/init.d/rc.board_sensors
rgbled #0 on I2C bus 2
WARN  [SPI_I2C] rgbled_ncp5623c: no instance started (no device on bus?)
WARN  [SPI_I2C] hmc5883: no instance started (no device on bus?)
WARN  [SPI_I2C] ms5611: no instance started (no device on bus?)
ms5611 #0 on SPI bus 1 (devid=0x3d)
mpu6000 #0 on SPI bus 1 (devid=0x21)
WARN  [SPI_I2C] mpu9250: no instance started (no device on bus?)
l3gd20 #0 on SPI bus 1 (devid=0x22)
lsm303d #0 on SPI bus 1 (devid=0x11)
WARN  [SPI_I2C] adis16448: no instance started (no device on bus?)
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS3
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [px4io] default PWM output device
INFO  [init] Mixer: /etc/mixers/AAERTWF.main.mix on /dev/pwm_output0
INFO  [pwm_out] instance: 0, MAX RATE: 100, default: 50, alt: 50
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1
ERROR [px4io] failed setting PWM rate on IO
ERROR [pwm] PWM_SERVO_SET_SELECT_UPDATE_RATE
ekf2 [485:100]
INFO  [logger] logger started (mode=all)
INFO  [logger] logging 49 topics from logger_topics.txt

NuttShell (NSH) NuttX-10.0.0
nsh> ERROR [ekf2] 0 - vehicle_magnetometer lost, generation 18 -> 25
INFO  [airspeed_selector] No airspeed sensor detected. Switch to non-airspeed mode.

```

Jaeyoung-Lim, 2021-07-17:
More debug outputs:
```
INFO  [SPI_I2C] instantiate failed (no device on bus 4 (devid 0x10000000)?)
WARN  [SPI_I2C] adis16448: no instance started (no device on bus?)
INFO  [SPI_I2C] instantiate failed (no device on bus 1 (devid 0x0)?)
INFO  [SPI_I2C] instantiate failed (no device on bus 1 (devid 0x0)?)
INFO  [SPI_I2C] instantiate failed (no device on bus 1 (devid 0x0)?)
INFO  [SPI_I2C] instantiate failed (no device on bus 1 (devid 0x0)?)
INFO  [SPI_I2C] instantiate failed (no device on bus 1 (devid 0x0)?)
INFO  [SPI_I2C] instantiate failed (no device on bus 1 (devid 0x0)?)
INFO  [SPI_I2C] instantiate failed (no device on bus 1 (devid 0x0)?)
INFO  [SPI_I2C] instantiate failed (no device on bus 1 (devid 0x0)?)
INFO  [SPI_I2C] instantiate failed (no device on bus 1 (devid 0x0)?)
INFO  [SPI_I2C] instantiate failed (no device on bus 1 (devid 0x0)?)
INFO  [SPI_I2C] instantiate failed (no device on bus 1 (devid 0x0)?)
INFO  [SPI_I2C] instantiate failed (no device on bus 1 (devid 0x0)?)
INFO  [SPI_I2C] instantiate failed (no device on bus 1 (devid 0x0)?)
INFO  [SPI_I2C] instantiate failed (no device on bus 1 (devid 0x0)?)
```

acfloria, 2021-07-26:
I tried to debug why the startup fails by adding some more debug outputs:

INFO  [adis16448] ADIS instantiate called
INFO  [adis16448] ADIS16448 Constructor: bus = 4
INFO  [adis16448] ADIS16448 Constructor: device = 268435456
INFO  [adis16448] ADIS16448 Constructor: bus_frequency = 2000000
INFO  [adis16448] ADIS instance created
INFO  [drivers__device] SPI Init called
INFO  [adis16448] Probe called
INFO  [adis16448] PROD_ID 0
INFO  [adis16448] SERIAL_NUM 0
INFO  [adis16448] LOT_ID1 0
INFO  [adis16448] LOT_ID2 0
INFO  [adis16448] unexpected PROD_ID 0x00
INFO  [adis16448] PROD_ID 0
INFO  [adis16448] SERIAL_NUM 0
INFO  [adis16448] LOT_ID1 0
INFO  [adis16448] LOT_ID2 0
INFO  [adis16448] unexpected PROD_ID 0x00
INFO  [adis16448] PROD_ID 0
INFO  [adis16448] SERIAL_NUM 0
INFO  [adis16448] LOT_ID1 0
INFO  [adis16448] LOT_ID2 0
INFO  [adis16448] unexpected PROD_ID 0x00
INFO  [drivers__device] probe failed
INFO  [adis16448] ADIS16448.init(): ret = -1
INFO  [adis16448] ADIS16448.init(): ret = not OK
INFO  [adis16448] SPI::init failed (-1)

It seems as if the SPI communication is not working because it is returning always 0. The hardware is fine. When flashing it with our old custom firmware the product ID is correctly read as 0x4040 and the driver is working.

acfloria, 2021-07-26:
I tested it with some older versions:

**1.11.0**
Same as above, returns all 0

**1.10.0**
The product ID gets detected but the measure fails:

INFO  [adis16448] ADIS16448 is detected ID: 0x4040, Serial: 0x1b
ADIS16448 on SPI bus 4 at 4 (1000 KHz)
ERROR [adis16448] measure failed
ERROR [adis16448] driver start failed

**1.9.0**
The Adis driver starts up and works.

closed 2021-07-30 by dagar