PX4/PX4-Autopilot #17875: armv7-m/arm_hardfault.c line: 134 task: mavlink_rcv_if0

Abner0907, 2021-07-09:
I have a similar problem like this. IT will crash/hardfault when running mavlink_rcv_if0;
https://github.com/PX4/PX4-Autopilot/issues/17063
and I follow the solution:  https://github.com/PX4/PX4-Autopilot/commit/599851a025ef30cd910ee4435cb97510bef2537e
but it still crash when connnect the qgroundcontrol.
Anyone can help me ? I am very appreciate.
BTW: I use the STM32H743 chip with my private board.
My error sceenshot: (edited) 
[image]

dagar, 2021-07-09:
How close are you to current master or v1.12?

Abner0907, 2021-07-09:
yes, I use the latest version: https://github.com/PX4/PX4-Autopilot/releases/tag/v1.12.0-rc1

lukegluke, 2021-07-09:
Recently encountered on such problem with CDev https://github.com/PX4/PX4-Autopilot/issues/17846. Isn't your case?

Abner0907, 2021-07-10:
@lukegluke Thank you. I check your commint https://github.com/PX4/PX4-Autopilot/pull/17847, It seems fix my problem. many thanks.

lukegluke, 2021-07-10:
Nice, glad it helps. Just in case, be aware that MAVlink will not connect again to device after it appears back https://github.com/PX4/PX4-Autopilot/issues/17846.

Abner0907, 2021-07-20:
@lukegluke Hi, Sorry to interrupt you again. My system it

Abner0907, 2021-07-20:
@lukegluke Hello, I have this issues even now I have add what have done #17847 .
Could you please help me give some info here: 
BTW: I rebase on the v1.12.0 stable version. This commint https://github.com/PX4/PX4-Autopilot/releases/tag/v1.12.0-rc1
``` 
nsh> WARN  [PreFlightCheck] Preflight Fail: Compass 0 uncalibrated
WARN  [PreFlightCheck] Preflight Fail: Accel 0 uncalibrated
INFO  [mavlink] Sarm_hardfault: Hard Fault:
arm_hardfault:   IRQ: 3 regs: 0x200155c4
arm_hardfault:   BASEPRI: 00000080 PRIMASK: 00000000 IPSR: 00000003 CONTROL: 00000000
arm_hardfault:   CFAULTS: 00008200 HFAULTS: 40000000 DFAULTS: 00000000 BFAULTADDR: 0000000b AFAULTS: 00000000
arm_hardfault: PANIC!!! Hard fault: 40000000
up_assert: Assertion failed at file:armv7-m/arm_hardfault.c line: 134 task: mavlink_rcv_if0
up_registerdump: R0: ffffffff 240012e0 20014710 ffffffff ffffffff 240012e4 20014710 20013305
up_registerdump: R8: 20015710 00000000 00000000 00000046 00000000 20015698 0802346d 0802443c
up_registerdump: xPSR: 01000000 BASEPRI: 00000080 CONTROL: 00000000
up_registerdump: EXC_RETURN: ffffffe9
up_dumpstate: sp:     24001478
up_dumpstate: IRQ stack:
up_dumpstate:   base: 24001500
up_dumpstate:   size: 00000200
up_dumpstate:   used: 00000178
up_stackdump: 24001460: 00000000 00001134 24001478 2400564c 20015a28 08029d55 00000080 00000000
up_stackdump: 24001480: 00000046 00000000 20015698 0802346d 0802443c 08157dc8 e000ed2c 08157dc8
up_stackdump: 240014a0: e000ed2c 00000000 00000003 20015710 00000000 00000000 00000046 080219f7
up_stackdump: 240014c0: 08157eaa 080216b9 40000000 00000000 0000000b 00000000 2400564c 08022f93
up_stackdump: 240014e0: 00000080 080210b1 00000080 200155c4 240012e4 20014710 20013305 080202d7
up_dumpstate: sp:     20015698
up_dumpstate: User stack:
up_dumpstate:   base: 20015a28
up_dumpstate:   size: 00001134
up_dumpstate:   used: 00000574
up_stackdump: 20015680: 00000000 00000000 00000000 00000000 00000001 00000000 240012e0 0802345f
up_stackdump: 200156a0: 20014710 08023443 240012e0 20014710 000000f0 08023279 2001b210 2001b210
up_stackdump: 200156c0: 00000000 08143313 00000000 00000000 00000000 00000000 00000000 2001b210
up_stackdump: 200156e0: 2001b210 00000000 20013305 00000000 00000000 00000000 00000017 0814238f
up_stackdump: 20015700: 2001b210 080bbdfb 2400566c 080233b3 2001b210 00000020 2400566c 2001b210
up_stackdump: 20015720: 20012c70 2001b210 20012c70 00000000 20013305 080bbf47 0000004f 20012c88
up_stackdump: 20015740: 200158b4 080bbfc5 00000000 060a0000 00000a01 00000000 00000000 00000000
up_stackdump: 20015760: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 20015780: 00000000 00000000 00000000 00000000 00000000 08000000 200158b4 20012c88
up_stackdump: 200157a0: 0000002b 080c699f 80000000 20012c70 20012c88 080c70bb deadbeef deadbeef
up_stackdump: 200157c0: 00000016 deadbeef 03c31b3c 00000000 00000000 20013428 deadbeef deadbeef
up_stackdump: 200157e0: deadbeef deadbeef deadbeef deadbeef deadbeef 00000003 00000101 00000000
up_stackdump: 20015800: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 002b2600
up_stackdump: 20015820: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 20015840: 00000000 00000000 03c32a1e 00000000 00000013 00000000 00000014 00000014
up_stackdump: 20015860: 00000000 00140000 00000000 00000000 deadbeef deadbeef deadbeef deadbeef
up_stackdump: 20015880: deadbeef deadbeef deadbeef deadbeef deadbeef deadbeef deadbeef deadbeef
up_stackdump: 200158a0: deadbeef deadbeef deadbeef deadbeef deadbeef 4ffe2b26 ff060000 00007e00
up_stackdump: 200158c0: 00000000 060a0000 00000a01 00000000 00000000 00000000 00000000 00000000
up_stackdump: 200158e0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 20015900: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 20015920: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 20015940: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 20015960: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 20015980: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 200159a0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 200159c0: 00000000 00000000 00002b26 00000000 00000000 00000000 20004cd0 00000000
up_stackdump: 200159e0: 00000000 20014710 200148a0 20011be0 20011c04 00000000 00000000 00000000
up_stackdump: 20015a00: 00000000 080c726f 080c7269 0803561b 00000000 00000000 00000000 00000000
up_stackdump: 20015a20: 00000000 00000000 00000060 80001150 20015a44 08024a49 00000000 00000002
up_taskdump: Idle Task: PID=0 Stack Used=332 of 728
up_taskdump: hpwork: PID=1 Stack Used=340 of 1268
up_taskdump: lpwork: PID=2 Stack Used=340 of 1620
up_taskdump: init: PID=3 Stack Used=2124 of 2932
up_taskdump: wq:manager: PID=4 Stack Used=396 of 1260
up_taskdump: mavlink_if1: PID=328 Stack Used=1756 of 2740
up_taskdump: mavlink_rcv_if1: PID=329 Stack Used=1308 of 4404
up_taskdump: logger: PID=402 Stack Used=2412 of 3644
up_taskdump: log_writer_file: PID=405 Stack Used=364 of 1172
[... 1 lines cut ...]
up_taskdump: wq:hp_default: PID=26 Stack Used=1276 of 1900
up_taskdump: wq:uavcan: PID=155 Stack Used=1828 of 3628
up_taskdump: dataman: PID=28 Stack Used=788 of 1204
up_taskdump: wq:lp_default: PID=30 Stack Used=1220 of 1924
[... cut ...]
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:0
[boot] Fault Logged on 2021-07-20-05:07:41 - Valid
[boot] There is a hard fault logged. Hold down the SPACE BAR, while booting to review!
sercon: Registering CDC/ACM se[hardfault_log] Fault Log info File No 4 Length 3177 flags:0x01 state:0
[hardfault_log] Fault Logged on 2021-07-20-05:07:41 - Valid
rial driver
sercon:[hardfault_log] Saving Fault Log file /fs/microsd/fault_2021_07_20_05_07_41.log
 Successfully registered the CDC/ACM serial driver

lukegluke, 2021-07-20:
I think firstly you need to track down exact place where hardfault occurs: go through up_stackdump addresses in flash (that starts with 0x080) from left to right. Follow this tutorial https://docs.px4.io/master/en/debug/gdb_debugging.html

Abner0907, 2021-07-20:
Thank you. I will have a debug to find the reason.

BTW: when I cherry-pick your comint and [this](https://github.com/PX4/PX4-Autopilot/commit/599851a025ef30cd910ee4435cb97510bef2537e). This never happen again by now.

lukegluke, 2021-07-20:
My commit was regarding hardfault on CDev disconnection when it used (by MAVlink for example) in custom build. But as it seemed to me, general px4 doesn't do such things, so if you are using regular px4, I don't think this case is related to you, you haven't describe what you are working with and on what links MAVlink is started.

Abner0907, 2021-07-21:
@lukegluke ,sorry late to reply. I have used a custom board and add a new prototype which used the STM32H743 chip.
My changed commit is here:  https://github.com/Abner0907/Firmware/tree/taloon1000/H7-v1.12.0-rc1
The crash is no happened again.

closed 2021-07-22 by Abner0907