PX4/PX4-Autopilot #17878: Wrong pwm value in disarm state

mengchaoheng, 2021-07-09:
**Describe the bug**
in release/1.12 branch:
When the aircraft is not unlocked, the output value of the servo remains at 900. This will turn the steering gear to the limit value or even destroy the steering gear

**To Reproduce**
Steps to reproduce the behavior:
1. make px4_sitl gazebo_plane
2. Observe the pwm output in the arm and disarm states respectively

**Expected behavior**
in v1.11.3, When the safety switch is disabled, the pwm value of the channel where the steering gear is located is the value obtained by the controller output through the mixer.

mengchaoheng, 2021-07-10:
in https://github.com/PX4/PX4-Autopilot/releases, it is sad that “Safety (switch) defaults to off (motors are disarmed, but servos / flaps can move)”，but now the servo can't move.

mengchaoheng, 2021-07-10:
I have found the reason. The reason why the pwm value is different from the expected one is that the description document (https://docs.px4.io/master/en/advanced_config/prearm_arm_disarm.html) on COM_PREARM_MODE and CBRK_IO_SAFETY is inconsistent with the actual situation. And the direct performance of v1.11 and v1.12 is different. It's best to actually check. It's time to change the documentation.

hamishwillee, 2025-11-06:
Closing. If needed, please reproduce on current version of PX4 - and as this indicates a docs issue, can you label it as Documentation (I will then see it)