PX4/PX4-Autopilot #17880: support for make px4_sitl_ctrlalloc gazebo_iris_ctrlalloc

ErcBunny, 2021-07-10:
## Describe the bug
1. The model for gazebo_iris_ctrlalloc is missing. Though it is easy to create one by copy&pasting and tweaking names, please fix this or the corresponding airframe file is a bit lonely to be there.
2. `make px4_sitl_ctrlalloc gazebo_iris_ctrlalloc -j8` and the drone doesn't take off on `commander takeoff`. This issue can be solved by changing the if condition of `ActuatorEffectivenessMultirotor::getEffectivenessMatrix` to `if (_parameter_update_sub.updated()){...}`. That is, by ignoring the bool variable `_updated` the effectiveness matrix can be updated as expected and therefore the issue no longer exists. If I do not do so, the effectiveness matrix is always zero. In fact, I don't see any logical problem with the original code at all. So I'm so confused right now.

## To Reproduce
1. Create a model for gazebo simulation.
2. `make px4_sitl_ctrlalloc gazebo_iris_ctrlalloc -j8`

## Screenshots
[image]

Jaeyoung-Lim, 2021-07-13:
@ErcBunny Fixed in https://github.com/PX4/PX4-SITL_gazebo/pull/781

ErcBunny, 2021-07-13:
@Jaeyoung-Lim Hi Jaeyoung, thanks for fixing the model problem. Do you have any thoughts about the problem related to  `if(_updated || _parameter_update_sub.updated())`?

ErcBunny, 2021-09-14:
> @Jaeyoung-Lim Hi Jaeyoung, thanks for fixing the model problem. Do you have any thoughts about the problem related to `if(_updated || _parameter_update_sub.updated())`?

Solved in #18200

closed 2021-09-14 by ErcBunny