PX4/PX4-Autopilot #17887: Offboard mode without gps

Carlosmct, 2021-07-12:
I´m working with PX4 with a Multicopter using offboard mode with ROS. I fly my drone with a fuse of gps and vision pose estimate with EKF2 as estimatior. Now I want fly without gps, using only the vision pose estimate. I configurate the EKF2_AID_MASK to use only the vision pose estimation, allow arming without gps in COM_ARM_WO_GPS and configure the EKF2_HGT_MODE to use the vision pose estimation. But now, when I try to switch to offboard mode, it doesn´t switch. 

[ INFO] [1626084483.441063658, 135.998000000]: Offboard enabled
INFO [tone_alarm] notify negative
[ WARN] [1626084470.224055975, 125.904000000]: CMD: Unexpected command 176, result 1
[ INFO] [1626084476.515195445, 130.722000000]: HP: requesting home position 

Thanks!

ecmnet, 2021-07-22:
Might be fixed with https://github.com/PX4/PX4-ECL/pull/1001

DavidePatria, 2022-02-04:
@Carlosmct  do you have any news regarding this issue?