PX4/PX4-Autopilot #17890: Error: make px4_sitl gazebo

graysondu, 2021-07-12:
~/github/PX4-Autopilot$ make px4_sitl_default gazebo
[0/4] Performing build step for 'sitl_gazebo'
[1/66] Building CXX object CMakeFiles/...plugin.dir/src/gazebo_uuv_plugin.cpp.o
FAILED: CMakeFiles/gazebo_uuv_plugin.dir/src/gazebo_uuv_plugin.cpp.o 
/usr/bin/ccache /usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -Dgazebo_uuv_plugin_EXPORTS -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-6.2 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -I/home/adu/github/PX4-Autopilot/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-9/gazebo/msgs -I/home/adu/github/PX4-Autopilot/mavlink/include -isystem /usr/include/opencv -I/home/adu/github/PX4-Autopilot/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/uuid -isystem /usr/include/x86_64-linux-gnu -O2 -g -DNDEBUG -fPIC   -Wno-deprecated-declarations -Wno-address-of-packed-member -I/usr/include/uuid -I/usr/include/x86_64-linux-gnu -std=gnu++1z -MD -MT CMakeFiles/gazebo_uuv_plugin.dir/src/gazebo_uuv_plugin.cpp.o -MF CMakeFiles/gazebo_uuv_plugin.dir/src/gazebo_uuv_plugin.cpp.o.d -o CMakeFiles/gazebo_uuv_plugin.dir/src/gazebo_uuv_plugin.cpp.o -c /home/adu/github/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_uuv_plugin.cpp
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
[2/66] Building CXX object CMakeFiles/...ir/src/gazebo_opticalflow_plugin.cpp.o
FAILED: CMakeFiles/gazebo_opticalflow_plugin.dir/src/gazebo_opticalflow_plugin.cpp.o 
/usr/bin/ccache /usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -Dgazebo_opticalflow_plugin_EXPORTS -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-6.2 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -I/home/adu/github/PX4-Autopilot/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-9/gazebo/msgs -I/home/adu/github/PX4-Autopilot/mavlink/include -isystem /usr/include/opencv -I/home/adu/github/PX4-Autopilot/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/uuid -isystem /usr/include/x86_64-linux-gnu -I/home/adu/github/PX4-Autopilot/Tools/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker/include -O2 -g -DNDEBUG -fPIC   -Wno-deprecated-declarations -Wno-address-of-packed-member -I/usr/include/uuid -I/usr/include/x86_64-linux-gnu -std=gnu++1z -MD -MT CMakeFiles/gazebo_opticalflow_plugin.dir/src/gazebo_opticalflow_plugin.cpp.o -MF CMakeFiles/gazebo_opticalflow_plugin.dir/src/gazebo_opticalflow_plugin.cpp.o.d -o CMakeFiles/gazebo_opticalflow_plugin.dir/src/gazebo_opticalflow_plugin.cpp.o -c /home/adu/github/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_opticalflow_plugin.cpp
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
^Cninja: build stopped: interrupted by user.
ninja: build stopped: interrupted by user.
^CMakefile:217: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Interrupt

Jaeyoung-Lim, 2021-07-12:
@graysondu Please follow the issue template in order to provide enough information of your issue

Jaeyoung-Lim, 2021-10-04:
Closing as original author of the issue did not come back

closed 2021-10-04 by Jaeyoung-Lim