PX4/PX4-Autopilot #17893: GPS driver not starting - PX4 crash

alessandrogenoni, 2021-07-13:
**Describe the bug**
After updating to 1.12.0, using a here2 GPS over UAVCAN, the software crashes when trying to start gps using the appropriate command. Even the test function of the GPS driver doesn't currently work 

**To Reproduce**
Steps to reproduce the behavior:
1. Start the board without serial GPS and serial GPS disabled
2. Start GPS from console
3. PX4 crashes and needs reboot

**Expected behavior**
The software should recognize the GPS automatically and start the drivers accordingly 

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=2ec694d0-2f99-42b4-a987-a354b9a1b468

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Tailsitter VTOL using Cube Orange

alessandrogenoni, 2021-07-13:
Even using the serial connection, in the pre-flight checks, it says that gps is disabled even if the command `gps status` reads normal values. Did this update break the gps driver?

dagar, 2021-07-13:
> the software crashes when trying to start gps using the appropriate command

Can you expand on this? Why are you running the `gps` command (serial driver) if trying to use UAVCAN GPS?

What does `uavcan status` show?

alessandrogenoni, 2021-07-13:
Ok, my bad. Didn't know that the `gps` drive handled only the serial connections. What handles the GPS over UAVCAN?
`uavcan status` shows the GPS connected and the compass is responding over UAVCAN too, so I guess that that part is fine

alessandrogenoni, 2021-07-14:
The gps actually works, but still shows as disabled in the pre-arm checklist in QGC. The satellite count is showing correctly and I get  a fix. On the other end I'm not sure the barometer is working. I've enabled it in uavcan and saved the parameters but i still get an error

closed 2021-07-16 by alessandrogenoni