PX4/PX4-Autopilot #17894: Use HITL to calibrate the compass in qgc

SamLin55587, 2021-07-13:
When  I  use Hardware In the Loop with PixHawk cube 2.1 , QGroundControl & Gazebo simulation and calibrate the Compass ,it doesn't have any response also happening in Accelerometer.
Following video is my situation.
Can someone help me？
Sincerely appreaciate

https://user-images.githubusercontent.com/83861951/125421663-ac966fc6-98ae-459b-a097-7ec61523ecfb.mp4

raghav-atd, 2021-07-16:
I am facing a very similar problem. Unable to calibrate the sensors as required by QGC in my HITL simulation on Gazebo. For obvious reasons, it is not possible to rotate and hold the quadcopter in the required positions in Gazebo.

Using Pixhawk Cube with PX4 software version 1.11 with the latest Daily Buid of QGC. We have written our own scripts that use MAVROS. We would like these scripts to run on an NVidia Jetson Nano to control the aircraft using offboard control. To test HITL first, this calibration problem is holding us back. I believe it restricts the drone from being set to Offboard state in QGC.

Any help will be greatly appreciated, thanks in advance!

JStoop97, 2021-07-17:
Yes I also had this problem and raised the issue [here](https://github.com/PX4/PX4-Autopilot/issues/17918). I am running 1.11.0, it could be this specific firmware but haven't tried with others yet. I circumvented the problem by using SIH instead which seems to run just fine.