PX4/PX4-Autopilot #17895: Add MAVLINK debug features

drtrigon, 2021-07-13:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

**Describe problem solved by the proposed feature**
When building additional hardware (e.g. a companion computer) that needs to interact with the FC and send MAVLINK messages to the FC or the GroundStation (e.g. via tunnel) it is not possible to figure out whether the messages do get sent or just dropped at all.
Example: https://discuss.px4.io/t/how-to-debug-mavlink-connections-and-messages/23159

**Describe your preferred solution**
A console or GUI tool that allows to inspect all MAVLINK messages traffic on the networks.

**Describe possible alternatives**
(no idea)

dagar, 2021-07-13:
Have you tried Wireshark?

https://mavlink.io/en/guide/wireshark.html

drtrigon, 2021-07-15:
Yes.

We have several issues:
1. wireshark seems to depend on the features of the network card in the machine and at least some of the wifi card seem not to be able to sniff all the network traffic properly
2. how to figure out what messages are sent if they do get sent to a specific machine/node and not all - what if the message is targeted for another machine/node and not a broadcast?

Any ideas?

Audifire, 2021-07-16:
Hi, I am a team member of this project and try to describe what’s the problem here. 

Here is the small setup we use. 

Companion PC (SysID 1 Component 25) <==> flight controller <==> GCS (SysID 255 Component 190) 

We try to send a MAVLink tunnel message from the companion PC via flight controller to the GCS (From System 1 Component 25 – To System 0 (Broadcast) Component 190). So basically, as I understood this should be a broadcast message and I should be able to monitor it on every interface. But I am not able. In the QGC in the MAVLink Inspector I cannot see a message type TUNNEL. I try to monitor via USB and via WiFi module. So now the important question. How can I monitor efficiently the package flow of MAVLink messages? In Linux I have the possibility to use tcpdump when I want to monitor network package flow. Is there something similar? How can I find out in this situation where the problem is? 

Thank you already for your help. 

Greetings, Audifire

ecmnet, 2021-07-22:
I did it the other way round:  FlightController <=> Companion <=> GCS. The companion handles the dispatching.