PX4/PX4-Autopilot #17900: [nxp_fmuk66-e] Radio channel detect and GPS broken on v1.12.0 stable

landonh12, 2021-07-13:
On v1.12.0 stable the NXP FMUK66-E has broken RC Channel detection and GPS. @jarivanewijk Tested these functions on v1.12.0 Beta 6, which was built before the series of commits regarding DMA on the FMUK66-E (5ea4b7dc9e17a0122714807d3432f5db86de6594). 

[image]

[image]

@davids5 @PetervdPerk-NXP @igalloway

davids5, 2021-07-13:
@landonh12 - Please test https://github.com/PX4/PX4-Autopilot/pull/17907

landonh12, 2021-07-14:
Tested - screenshots @ https://github.com/PX4/PX4-Autopilot/pull/17907#issuecomment-880152987