PX4/PX4-Autopilot #17906: Cannot fit mag #0

Ziov, 2021-07-13:
Using a pixhawk 2.4.8 and latest firmware using qgroundcontrol, trying to calibrate mag but keep getting an error saying "cannot fit mag #0"

I think there are 3 mags two in board and one in gps but keep getting this error.

Anyone know how to fix it?

dagar, 2021-07-13:
Could you enable logging from boot (SDLOG_MODE 1) and try again? We now have logging during calibration (`mag_worker_data`).

Ziov, 2021-07-13:
> Could you enable logging from boot (SDLOG_MODE 1) and try again? We now have logging during calibration (`mag_worker_data`).

Do I download the logs using qgroundcontrol?

dagar, 2021-07-13:
You can use qgroundcontrol to transfer the log or get it directly from the SD card.