PX4/PX4-Autopilot #17911: 1.12 Release everything great except Weather Vane

taileron, 2021-07-14:
**Describe the bug**
since the beginning of 2021 Weather Vane no longer works and also in the 1.12.0 release. Is there a trick to activate WV?

**To Reproduce**
Convergence Airframe
Forward Thrust enabled
Enough wind

**Expected behavior**
WV works and turns the aircraft into the wind from a slight roll inclination.

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=a8f081af-ee68-42a0-81e4-71768cec0ed2

RomanBapst, 2021-07-14:
@taileron Is seems like weathervane support for position control flight mode was removed, I can't see it anymore.
It should still work in RTL though.

@MaEtUgR Can you comment on this?

taileron, 2021-07-14:
I love the matrix as implemented for VT_FWD_THRUST_EN.
WV would be an extreme increase in flight safety for pos mode as soon as the aircraft is hovering on the spot in strong winds. 
Great would be WV in pos mode which would be turned off for manual forward movements.

taileron, 2021-07-26:
@MaEtUgR I just don't fully understand the logic in the code yet. Lateral movements are taken into account, but where can I find that WV is only activated on rtl? This could also be the reason why during the last descent of the rtl procedure for landing, the aircraft turns around its axis when there is only little wind. How can I check if or when WV is active in flight and if it is working and set correctly?

VTOLDavid, 2021-07-29:
> 
> 
> @taileron Is seems like weathervane support for position control flight mode was removed, I can't see it anymore.
> It should still work in RTL though.
> 
> @MaEtUgR Can you comment on this?

Why it was removed? It is useful in position mode

dagar, 2021-08-08:
We need to implement the yaw handler for FlightTaskManualAcceleration. 

https://github.com/PX4/PX4-Autopilot/blob/c8366e736e7e458a4d573bcf0d49c78ba81ac2d1/src/modules/flight_mode_manager/tasks/ManualPosition/FlightTaskManualPosition.hpp#L55-L58

https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/flight_mode_manager/tasks/ManualAcceleration/FlightTaskManualAcceleration.hpp

taileron, 2021-08-10:
I made this change in V1.12.1 and WV now works very well in pos mode. 
I can provide logs later.
I will test the original pr next.

closed 2021-08-12 by dagar