PX4/PX4-Autopilot #17927: [imuConsistencyCheck] Accel/gyro health checks broken

ThomasRigi, 2021-07-16:
**Describe the bug**
The `sensor_status_imu` topic only reports one healthy acc/gyro sensor even if multiple are healthy, and it is seemingly random if it's the first or the second that is reported as healthy in [SYS_STATUS](https://mavlink.io/en/messages/common.html#SYS_STATUS)' `onboard_control_sensors_health` field. 

The failing check is located here: https://github.com/PX4/PX4-Autopilot/blame/master/src/modules/commander/Arming/PreFlightCheck/checks/imuConsistencyCheck.cpp#L58, which was changed by @dagar's PR https://github.com/PX4/PX4-Autopilot/pull/15681.

I think the issue lies somewhere in the sensor selection module, see here for a starting point: https://github.com/PX4/PX4-Autopilot/blame/master/src/modules/sensors/voted_sensors_update.cpp#L410. 

Analogous for gyro, of course.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on a Pixhawk 4 running v1.12.0 (or master)
2. in a mavlink shell, type `listener sensors_status_imu` 
3. only one field of `accel_healthy` and `gyro_healthy` is true.
4. also check the bitmap [SYS_STATUS](https://mavlink.io/en/messages/common.html#SYS_STATUS)' `onboard_control_sensors_health`. Depending on the `sensor_selection` topic it's either the first or the 2nd gyro/accel that is reported as healthy and the other as not healthy.

**Expected behavior**
Both sensors should be reported as healthy when they are healthy.

Add screenshots to help explain your problem.
Output of QGC's mavlink shell, reboot in the middle:
[image]

**Drone (please complete the following information):**
- Observed on a Pixhawk 4 running v1.12.0, which was previously working fine on v1.11.3. 

**Additional context**
Add any other context about the problem here.

dagar, 2021-07-16:
Is multi-ekf active?
 - `ekf2 status`, `param show EKF2_MULTI*`, etc?

With multi-ekf enabled this decision effectively shifts from `sensors` to the estimator selector. If that's the case then I'll try to populate these flags differently (they're mainly for casual inspection).

ThomasRigi, 2021-07-16:
Yes, they're active from as much as I understand. We want to use these flags to monitor the drone's health on the onboard computer and to know if we can arm the drone without actually having to send an arm command. For now we set our check that if one is good we consider the drone healthy but it would be nice if the check actually worked for both IMUs.

[image]
[image]

dagar, 2021-07-19:
See https://github.com/PX4/PX4-Autopilot/pull/17946.

closed 2021-07-23 by dagar