PX4/PX4-Autopilot #17928: orangecube version mismach 1.12 and simulation battery error

gendebeat, 2021-07-16:
Hello im triying to use orangecube with latest firmware 1.12. i need to use simulation too.
(yes i know this board is "special"... i can only upload firmware with missionplanner, no QGC)

When i download latest px4 for this board, QGS says that the firmware version is 1.11.2 !!
Of couse, with this binary, the simulation run with no problem...

When i compile source code for 1.12.. surprise, i can run in simulation but no arm because baterry error. 
maybe some problems with this problematic board?

thanks

karthiksoma-atd, 2021-07-17:
@gendebeat  I am having similar problems, were you able to calibrate your sensors and use the offborad mode in hitl?

gendebeat, 2021-07-19:
@karthiksoma-atd my sensors are calibrate previously. I activate SIH in QGC and cant arm because some battery error. I not use offboard mode, I only try to simulate a plan path with the orangecube and the lastest version.
With blackcube have no problem to simulate.

gendebeat, 2021-07-19:
ok, maybe in this version the simulation mode is diferent.  i need to use CBRK_SUPPLY_CHK to ignore the battery. now i can arm the vehicle

closed 2021-07-19 by gendebeat