PX4/PX4-Autopilot #17930: Which parameter is the coefficient of torque (Cq)?

kadhirumasankar, 2021-07-16:
I am modifying one of the SDFs in the PX4 firmware using coefficient of thrust and torque (Ct and Cq) values found experimentally. I realized that the parameter for Ct is `motorConstant`, which can be seen in its usage [here](https://github.com/PX4/PX4-SITL_gazebo/blob/1f3f1b1dec5a31c2c400f141fc344f11f68659c9/src/gazebo_motor_model.cpp#L201)

Which parameter is the coefficient of torque, Cq? Thanks in advance.

Jaeyoung-Lim, 2021-07-17:
@kadhirumasankar Since you are looking at the code already, you can see how torque is being computed here: https://github.com/PX4/PX4-SITL_gazebo/blob/1f3f1b1dec5a31c2c400f141fc344f11f68659c9/src/gazebo_motor_model.cpp#L250

kadhirumasankar, 2022-05-20:
I got more information from @mvernacc's document Gazebo Motor & Propeller Model Notes (https://github.com/mvernacc/gazebo_motor_model_docs/blob/master/notes.pdf)

closed 2022-05-20 by kadhirumasankar