PX4/PX4-Autopilot #17936: make jmavsim stuck

broke99, 2021-07-18:
**After running this** 
isohaib@ubuntu:~/PX4-Autopilot$ make px4_sitl jmavsim
[0/1] cd /home/isohaib/PX4-Autopilot/b...b/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/isohaib/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
world: none
src_path: /home/isohaib/PX4-Autopilot
build_path: /home/isohaib/PX4-Autopilot/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/home/isohaib/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/isohaib/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/isohaib/PX4-Autopilot"/test_data
Creating symlink /home/isohaib/PX4-Autopilot/build/px4_sitl_default/etc -> /home/isohaib/PX4-Autopilot/build/px4_sitl_d

_____  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
INFO  [parameters] BSON document size 447 bytes, decoded 447 bytes
[param] Loaded: eeprom/parameters_10016
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
Buildfile: /home/isohaib/PX4-Autopilot/Tools/jMAVSim/build.xml

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
INFO  [simulator] Simulator connected on TCP port 4560.
Init MAVLink
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [ekf2] starting instance 0, IMU:0 (1310988), MAG:0 (197388)
INFO  [ekf2] starting instance 1, IMU:1 (1310996), MAG:0 (197388)
INFO  [ekf2] starting instance 2, IMU:2 (1311004), MAG:0 (197388)
INFO  [ekf2] starting instance 3, IMU:0 (1310988), MAG:1 (197644)
INFO  [ekf2] starting instance 4, IMU:1 (1310996), MAG:1 (197644)
INFO  [ekf2] starting instance 5, IMU:2 (1311004), MAG:1 (197644)
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2021-07-18/12_38_21.ulg
INFO  [logger] Opened full log file: ./log/2021-07-18/12_38_21.ulg
INFO  [px4] Startup script returned successfully
pxh> INFO  [tone_alarm] home set
INFO  [tone_alarm] notify negative`

**The simulator starts but the process gets stuck at this INFO [tone_alarm] notify negative and I cannot execute the commander takeoff**

mrpollo, 2021-08-04:
Does the Java window open for you @broke99? What OS are you running, and which release (commit) are you building?

potaito, 2021-08-05:
@broke99 just some context. `INFO [tone_alarm] notify negative` is not an error, but this tune is always played when the home position is set. So far your output looks normal. In what way can you not execute `commander takeoff`? Can you show the output of that as well?

broke99, 2021-08-05:
Hi, I left that build while ago and shifted to ardupilot and I do not have any screenshot of the output for **commander takeoff** . As far as I remember I got something like landing detected and some values in the terminal and no response from the quadcopter in the jmavsim window. Btw thanks for your concern.

potaito, 2021-08-05:
Okay that's good. That means that in general your sim is working, but the drone is not taking off for some reason (like no RC). You will have to retry that and share the console output before we can help further. There could be many reasons why the takeoff is not working.

NicolassFelipe, 2021-08-24:
@potaito Hello, im having this same problem with "make px4_sitl jmavsim" and "make px4_sitl gazebo", im opening a new issue in a few moments, would you please check it out? ty!.

potaito, 2021-08-25:
@broke99 please re-open if this is still an issue and describe in more detail what exactly is not working. To me it's still not clear, since your console output looks correct.

closed 2021-08-25 by potaito