PX4/PX4-Autopilot #17943: Debug the stm32h743's bootloader issue

Abner0907, 2021-07-19:
Hi, I want to ask a question about how to debug the bootloader.
1: Now I can use the console function 'showprogress()' to debug. But this is very slow for me, I need flash the firmware again and again. And the worst is:   before [jump to app](https://github.com/PX4/PX4-Autopilot/blob/master/platforms/nuttx/src/bootloader/common/bl.c#L399), there are some disable system time, it can not console the progress, that means showprogress() is  invalid . This is very bad

2: I also have used the JLINK to debug the bootloader  with *.elf file. I found this is  very convenient, but sometimes i found that the program executed or breakpoint is not the program really want to do. (I can not dig out the real reason ). 
I am very appreciated that someone can let me the other way to debug the H7 bootloader

MaEtUgR, 2021-07-19:
@julianoes @davids5 ?

davids5, 2021-07-19:
Do not use the console. It will change the timing. Use the Jlink. (and [GNU Eclipse](https://eclipse-embed-cdt.github.io/) or VSCode) Use load `symbol-file <path to elf>`  select the application or the bootloader elf. Then you can set breakpoints in either.

Abner0907, 2021-07-19:
Thank you, Davids. You means the 'Cortex-Debug'  tool in vscode ?

davids5, 2021-07-19:
> Thank you, Davids. You means the 'Cortex-Debug' tool in vscode ?

I use Eclipse, so I do not know the tool setup for VS code.

Abner0907, 2021-07-19:
Thank you all the same. Davids. I will have a try.

Abner0907, 2021-07-19:
It is ok for me now, I think this issues can be closed. Thank you davids and Matthias

closed 2021-07-19 by Abner0907