PX4/PX4-Autopilot #17951: Thrust curve parameter cannot be changed

serday, 2021-07-20:
When configuring PIDs, you need to change thrust curve or THR_MDL_FAC  parameter from 0 to 1. firmware 1.12 does not allow you to do this. Firmware 1.11 allows you to do this.

dagar, 2021-07-21:
`THR_MDL_FAC` is definitely still present in v1.12. 

What are you seeing?

ebtrac, 2021-07-21:
Actually, it can be changed. The real issue is that you can only set `THR_MDL_FAC` to be 0 or 1. If you try to give it a value of say, 0.25, it will not change.

serday, 2021-07-21:
Ok, thanks. But how its must working? 
I see that "The mapping from motor control signals (e.g. PWM) to expected thrust is linear by default — setting THR_MDL_FAC to 1 makes it quadratic. Values in between use a linear interpolation of the two. Typical values are between 0.3 and 0.5."
from here - https://docs.px4.io/master/en/config_mc/pid_tuning_guide_multicopter.html
May be it is not important parameter

ebtrac, 2021-07-22:
I have figured out why this is happening. My version of QGroundControl was outdated. If your QGroundControl version doesn't say v4.1.3, you should uninstall it and download the latest release from github.

Here is what it should look like if you have the right version:
[image]

Download the latest version from here:

serday, 2021-07-23:
Thanks, you really helped. I will follow the new versions

closed 2021-07-23 by serday