PX4/PX4-Autopilot #17953: STM32H7's PA0 Can not read ADC

Abner0907, 2021-07-20:
@davids5 ,  Now I have a issue about the PA0, I can not read the adc value or read a random value, it will show a random value in PA0/CHANNEL 16 channel when I type `listener adc_report`. I can read the PA1 and PA3's adc value.
The datasheet said it has a PA0_C. I can not confirm whether this is the reason I can not read the adc. Maybe I have config the PA0 wrong? I am very appreciated that you can give me some notes. many thanks.
[image]

##BTW:## I have check the hardware. it is ok. And I modify the hardware to PA1. The adc is also ok.

davids5, 2021-07-20:
Also the schematic would help

Abner0907, 2021-07-21:
@davids5 Hello, the configure is here: https://github.com/Abner0907/Firmware/tree/taloon1000/H7-v1.12.0-rc1/boards/codev/dp1000-v2

The schematic's link is here: https://github.com/CodevDynamics/Opensource_Hardware/blob/main/DP_FC_SCH.pdf

FYI: In the schematic, I have use the STM32H743VIT6 for the chip is stm32f765vi, I have seen the pin definition is the same.

Abner0907, 2021-07-21:
@davids5  I can read channel 17 but can not read channel 16. So for hot fix, I jump a line to Channel 17(PA1) The configuration in my code.

davids5, 2021-07-21:
Stop in the debugger and and look at the GPIOA register set (GPIOA_MODER etal). See if it is still in analog mode? I would bet that something in the config is overwriting it. 

Also you have a CONFIG_ARCH_CHIP_STM32H743II (176 pin) set in the defconfig You would need to check the pinmap and see if it is it the same at the 100 pin part.

davids5, 2021-07-21:
Also, are the _C pins even bonded out on the V package? 

See my notes about the SYSCFG_PMCR default state.

https://github.com/PX4/PX4-Autopilot/blob/master/boards/px4/fmu-v6x/src/board_config.h#L154-L158

But I do not think it applies in the V package.

Abner0907, 2021-07-21:
@davids5 
**1:** 
>Also you have a CONFIG_ARCH_CHIP_STM32H743II (176 pin) set in the defconfig You would need to check the pinmap and see if it is it the same at the 100 pin part.

I have seen the definition in the datasheet. I think the pinmap is ok.

[image]

2: 
> Also, are the _C pins even bonded out on the V package?

Yes, I also think the _C pins is not on the V package.

3: 
> See my notes about the SYSCFG_PMCR default state.

I have read this register's value. The value is 0x3000000.

davids5, 2021-07-21:
@Abner0907 What do the GPIO regs show?

Abner0907, 2021-07-21:
@davids5 
I add a 6s battery and I measured it, the voltage is 25.3v.
The PA0 gpio's pin will show a random value between 6500-8500, the value will change all the time.

[image]

Abner0907, 2021-07-21:
My real voltage now is 22.03v. What is battery show is:
[image]

davids5, 2021-07-21:
dump the value in GPIOA MODER. Is is set to default? Check the GPIOA AFRL

[image]

Abner0907, 2021-07-21:
@davids5 
J-Link>mem32 0x58020020 1    (GPIOA AFRL)
58020020 = 55500000 

J-Link>mem32 0x58020000 1 (GPIOA MODER)
58020000 = 6A83A9FC

Abner0907, 2021-07-21:
@davids5 
Just now I have modify the GPIOA MODER's value to 6A83A9FF. The value can be detected correctly.
So the PA0's mode is be modified or changed by somewhere?

davids5, 2021-07-21:
Set a HW breakpoint on writes to MODER, then look up the stack.

PA maps to 00 and so does Pin 0 So it  anything that is broken passing a 0 to GPIO config can overwrite.  PA0...fun....fun...fun :)

Abner0907, 2021-07-22:
@davids5 
Interesting, Each time when I run this line, the `GPIOA->MODER->MODER0[1:0]` will change from '0b11' to `ob00`
[image]

When I comment out this line, the register's value will not change. 

It seems that `(GPIO_OTGFS_VBUS & GPIO_PUPD_MASK) | GPIO_PULLDOWN)` remap a wrong pin/input ?

Abner0907, 2021-07-22:
@davids5 
Hi, I found the reason, I have wrong configure the usb. define the vbus as below:
```
#define GPIO_OTGFS_VBUS         /* PA9 */ (GPIO_INPUT|GPIO_PULLDOWN|GPIO_SPEED_100MHz|GPIO_PORTA|GPIO_PIN9)
```
I configured it as
```
	// stm32_configgpio((GPIO_OTGFS_VBUS & GPIO_PUPD_MASK) | GPIO_PULLDOWN);
```
It should be configured as:
```
stm32_configgpio(GPIO_OTGFS_VBUS);

```

Abner0907, 2021-07-22:
@davids5 many thanks you for your help. It is very usefully.

closed 2021-07-22 by Abner0907