PX4/PX4-Autopilot #17962: stuck on sitl gazebo issue

freewhale98, 2021-07-22:
I'm trying to run a SITL simulation but having issues when building the code. I use Linux 18.04 and had this issue when I tried to build. 

```
gps@GPS:~/PX4-Autopilot$ make px4_sitl_default gazebo
-- PX4 version: v1.12.0-beta5-278-g9d3e18dbe1
-- PX4 config file: /home/gps/PX4-Autopilot/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 5.5.0
-- The C compiler identification is GNU 5.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-569-g71fc1b8
-- ROMFS: ROMFS/px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gps/PX4-Autopilot/build/px4_sitl_default
[0/817] git submodule src/drivers/gps/devices
[1/817] git submodule src/lib/ecl
[2/817] git submodule mavlink/include/mavlink/v2.0
[6/817] git submodule Tools/sitl_gazebo
[15/817] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- The C compiler identification is GNU 5.5.0
-- The CXX compiler identification is GNU 5.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++17 standard
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
[... 145 lines cut ...]
                     ^
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[533/817] Building CXX object src/modules/local_position_estimator/CMakeFiles/modules__local_position_estimator.dir/sensors/baro.cpp.o
FAILED: src/modules/local_position_estimator/CMakeFiles/modules__local_position_estimator.dir/sensors/baro.cpp.o 
/usr/bin/ccache /usr/bin/c++  -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -DMODULE_NAME=\"local_position_estimator\" -DPX4_MAIN=local_position_estimator_app_main -D__CUSTOM_FILE_IO__ -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/px4/sitl/src -I../../platforms/posix/src/px4/common/include -I. -Isrc/lib -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -I../../platforms/posix/include -Iexternal/Install/include -I../../src/lib/ecl -O2 -g -DNDEBUG -fPIC   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -std=gnu++14 -MD -MT src/modules/local_position_estimator/CMakeFiles/modules__local_position_estimator.dir/sensors/baro.cpp.o -MF src/modules/local_position_estimator/CMakeFiles/modules__local_position_estimator.dir/sensors/baro.cpp.o.d -o src/modules/local_position_estimator/CMakeFiles/modules__local_position_estimator.dir/sensors/baro.cpp.o -c ../../src/modules/local_position_estimator/sensors/baro.cpp
In file included from ../../src/lib/matrix/matrix/math.hpp:9:0,
                 from ../../src/lib/mathlib/math/Functions.hpp:45,
                 from ../../src/lib/mathlib/math/filter/LowPassFilter2p.hpp:41,
                 from ../../src/lib/controllib/BlockDelay.hpp:48,
                 from ../../src/lib/controllib/blocks.hpp:42,
                 from ../../src/modules/local_position_estimator/sensors/../BlockLocalPositionEstimator.hpp:7,
                 from ../../src/modules/local_position_estimator/sensors/baro.cpp:1:
../../src/lib/matrix/matrix/Matrix.hpp: In function ‘bool matrix::inv(const matrix::SquareMatrix<Type, M>&, matrix::SquareMatrix<Type, M>&, size_t) [with Type = float; long unsigned int M = 1ul; size_t = long unsigned int]’:
../../src/lib/matrix/matrix/Matrix.hpp:99:21: error: array subscript is above array bounds [-Werror=array-bounds]
         return _data[i][j];
                     ^
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[539/817] Building CXX object src/modu...ce.dir/geofence_breach_avoidance.cpp.o
ninja: build stopped: subcommand failed.
Makefile:224: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1

```

julianoes, 2021-07-23:
```
-- The CXX compiler identification is GNU 5.5.0
```

Why are you on GCC 5.5.0? From what I know, Ubuntu 18.04 comes with GCC 7.5.0 by default. I believe this might be the reason why it does not build.

Amna-Smaoui, 2022-04-27:
I have the same problem and I can't fix it 
[image]

Amna-Smaoui, 2022-04-27:
> @Amna-Smaoui <https://github.com/Amna-Smaoui> If you look at the error
> output, I believe your error is not the same as the original issue.

I did my best to solve it and I do a lot of researches but still I got the
same error

julianoes, 2022-05-15:
@Amna-Smaoui you should not be running the build with `sudo` as far as I know.

chasadch, 2024-11-01:
~/PX4-Autopilot/Tools/simulation/gazebo-classic$ git clone git://github.com/PX4/PX4-SITL_gazebo-classic.git sitl_gazebo-classic
Cloning into 'sitl_gazebo-classic'...
remote: Enumerating objects: 9667, done.
remote: Counting objects: 100% (111/111), done.
remote: Compressing objects: 100% (75/75), done.
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
asad@asad:~/PX4-Autopilot/Tools/simulation/gazebo-classic$ git submodule sync --recursive && git submodule update --init --recursive
Synchronizing submodule url for '../flightgear/flightgear_bridge'
Synchronizing submodule url for 'sitl_gazebo-classic'
Synchronizing submodule url for '../gz'
Synchronizing submodule url for '../jmavsim/jMAVSim'
Synchronizing submodule url for '../jsbsim/jsbsim_bridge'
Synchronizing submodule url for '../../../boards/modalai/voxl2/libfc-sensor-api'
Synchronizing submodule url for '../../../platforms/nuttx/NuttX/apps'
Synchronizing submodule url for '../../../platforms/nuttx/NuttX/nuttx'
Synchronizing submodule url for '../../../src/drivers/actuators/vertiq_io/iq-module-communication-cpp'
Synchronizing submodule url for '../../../src/drivers/cyphal/legacy_data_types'
Synchronizing submodule url for '../../../src/drivers/cyphal/libcanard'
Synchronizing submodule url for '../../../src/drivers/cyphal/public_regulated_data_types'
Synchronizing submodule url for '../../../src/drivers/gps/devices'
Synchronizing submodule url for '../../../src/drivers/uavcan/libdronecan/dsdl'
Synchronizing submodule url for '../../../src/drivers/uavcan/libdronecan/libuavcan/dsdl_compiler/pydronecan'
Synchronizing submodule url for '../../../src/lib/cdrstream/cyclonedds'
Synchronizing submodule url for '../../../src/lib/cdrstream/rosidl'
Synchronizing submodule url for '../../../src/lib/crypto/libtomcrypt'
Synchronizing submodule url for '../../../src/lib/crypto/libtommath'
Synchronizing submodule url for '../../../src/lib/crypto/monocypher'
Synchronizing submodule url for '../../../src/lib/events/libevents'
Synchronizing submodule url for '../../../src/lib/heatshrink/heatshrink'
Synchronizing submodule url for '../../../src/modules/mavlink/mavlink'
Synchronizing submodule url for '../../../src/modules/mavlink/mavlink/pymavlink'
Synchronizing submodule url for '../../../src/modules/uxrce_dds_client/Micro-XRCE-DDS-Client'
Synchronizing submodule url for '../../../src/modules/zenoh/zenoh-pico'
Cloning into '/home/asad/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic'...
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
fatal: clone of 'https://github.com/PX4/PX4-SITL_gazebo-classic.git' into submodule path '/home/asad/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic' failed
Failed to clone 'Tools/simulation/gazebo-classic/sitl_gazebo-classic'. Retry scheduled
Cloning into '/home/asad/PX4-Autopilot/Tools/simulation/jmavsim/jMAVSim'...
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
fatal: clone of 'https://github.com/PX4/jMAVSim.git' into submodule path '/home/asad/PX4-Autopilot/Tools/simulation/jmavsim/jMAVSim' failed
Failed to clone 'Tools/simulation/jmavsim/jMAVSim'. Retry scheduled
Cloning into '/home/asad/PX4-Autopilot/platforms/nuttx/NuttX/apps'...
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
fatal: clone of 'https://github.com/PX4/NuttX-apps.git' into submodule path '/home/asad/PX4-Autopilot/platforms/nuttx/NuttX/apps' failed
Failed to clone 'platforms/nuttx/NuttX/apps'. Retry scheduled
Cloning into '/home/asad/PX4-Autopilot/platforms/nuttx/NuttX/nuttx'...
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
fatal: clone of 'https://github.com/PX4/NuttX.git' into submodule path '/home/asad/PX4-Autopilot/platforms/nuttx/NuttX/nuttx' failed
Failed to clone 'platforms/nuttx/NuttX/nuttx'. Retry scheduled
Cloning into '/home/asad/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic'...
error: RPC failed; curl 92 HTTP/2 stream 0 was not closed cleanly: CANCEL (err 8)
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
fatal: clone of 'https://github.com/PX4/PX4-SITL_gazebo-classic.git' into submodule path '/home/asad/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic' failed
Failed to clone 'Tools/simulation/gazebo-classic/sitl_gazebo-classic' a second time, aborting
asad@asad:~/PX4-Autopilot/Tools/simulation/gazebo-classic$

chasadch, 2024-11-01:
sir kindly help me

julianoes, 2024-11-04:
@chasadch your issue is different, so please don't necrobump this issue.

Also, looks like something with your network connection might not be working properly:

```
 fatal: the remote end hung up unexpectedly
```

closed 2024-11-04 by julianoes

asqapro, 2025-04-11:
@friedpotato69 This commit fixes the issue: https://github.com/PX4/PX4-Autopilot/pull/21738