PX4/PX4-Autopilot #17969: EKF2_AID_MASK Unable to Set vision position fusion and vision yaw fusion

WaffleWolfie, 2021-07-23:
Hi all,

I have encountered an issue where I have tried to relay data from an Optitrack system to mavros. One of the procedures for the usage of EKF2 is to set the EKF2_AID_MASK to do vision position fusion and vision yaw fusion. 

The procedure to set the EKF2_AID_MASK is available here: https://dev.px4.io/v1.9.0_noredirect/en/ros/external_position_estimation.html

Upon checking the boxes and rebooting the PX4 as per the instructions, I am still unable to just check the 2 boxes. Instead upon opening the parameters, other parameters were also automatically checked. I am still not sure if this is what has been causing the problem where I have managed to relay the signal from the optitrack to the topic /mavros/vision_pose/pose. 
However, there is nothing published to the topic /mavros/local_position/pose even though mavros is also subscribed and publishing to the topic /mavros/local_position/pose.
If there is anyone that can help me it would be greatly appreciated! 
[image]
[image]

Thanks a bunch!

RossFrancies, 2021-07-23:
Hi, I also get the same EKF2_AID_MASK behavior but from the EKF2_AID_MASK menu you can also try using > advanced settings > manual input and then set this number to your desired fusion.

Best Regards, Ross

dagar, 2021-07-23:
I believe this is a QGC bug. Let me take a look.

As a short term work around you can set the parameter value directly (nsh> param set EKF2_AID_MASK 24).

WaffleWolfie, 2021-07-24:
Hi RossFrancies and Daniel Agar, 

I would like to first, thank you both for your help! I really appreciate it!

I have tried the solution RossFrancies recommended, but it did not seem to work for me. As I still see that it is still unable to set to the desired parameters.

Daniel Agar, I have yet to try the solution you have mentioned, but I will get to it the following Monday. I will update in the comments on the situation once I have tried out the solution.

Sincerely,
Clarence

WaffleWolfie, 2021-07-27:
Hi all,

Just an update, I have tried the solution provided by Daniel Agar. I have managed to achieve lift off. Although not a really controlled one as the drone tend to lift off and then yaw out of control, or lift off and then pitch out of control. I believe the issue is something to do with my other codes. Thank you for your help everyone!

Regards
Clarence