PX4/PX4-Autopilot #17972: ICM42688P SPI Not Working with RPi3!

Pskoog3, 2021-07-23:
I bought the following breakout board for ICM42688P and unable to get the SPI interface working with Raspberry Pi 3.
https://www.notwired.co/products/detail/nwmoticm42688p-notwired-co/646130/
The pull-up resistors have been removed for SPI as suggested in the attached file.

Although it is detected on SPI bus but failed to read the sensor. Have a look at the attached log. What could possibly be wrong?

[image]
[image]

dagar, 2021-07-23:
Can you compile the icm42688p with debug?

Add `-DDEBUG_BUILD` to the COMPILE_FLAGS here.

https://github.com/PX4/PX4-Autopilot/blob/af50e25647aa5c2051bb09378bc9b2aaaaeb2a07/src/drivers/imu/invensense/icm42688p/CMakeLists.txt#L37

Pskoog3, 2021-07-24:
Issue resolved. The problem was with jumper wires on the test bench.

closed 2021-07-24 by Pskoog3