PX4/PX4-Autopilot #17992: ninja: error: unknown target when making jsbsim targets

zimmy87, 2021-07-29:
**Describe the bug**
I'm trying to follow the docs at http://docs.px4.io/master/en/simulation/jsbsim.html but I can't get any of the make commands to work due to various errors outputted about unknown targets

**To Reproduce**
Steps to reproduce the behavior:
1. Install the usual Development Environment on Ubuntu LTS / Debian Linux.
1. Install a JSBSim release from the release page
    (opens new window):
    `dpkg -i JSBSim-devel_1.1.0.dev1-<release-number>.bionic.amd64.deb`
1. Run `make px4_sitl jsbsim`

**Expected behavior**
PX4 starts

**Log Files and Screenshots**
make command outputs error:

```
ninja: error: unknown target 'jsbsim', did you mean 'jmavsim'?
make: *** [Makefile:202: px4_sitl] Error 1
```

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.: PX4 SITL
- Photo of the IMU / autopilot setup if possible.

**Additional context**
This is happening on Ubuntu 20.04 with both master and v1.11.3

Sitin, 2023-06-26:
I have the same problem. Any update on this issue?

julianoes, 2023-10-19:
Do you have all the dependencies installed?

You need to add a couple of `message(STATUS blabla)` debug statements in https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/simulation/simulator_mavlink/sitl_targets_jsbsim.cmake to figure out what is wrong.