PX4/PX4-Autopilot #17993: Why does PX4 wait for simulator connection after starting jsbsim_quadrotor_x configuration

zimmy87, 2021-07-29:
**Describe the bug**
I'm confused about what the expected behavior is after making PX4 with the jsbsim_quadrotor_x configuration. I would expect that PX4 starts with JSBSim connected as the simulator, but PX4 still says its waiting for a connection from a simulator. Currently following the documentation at https://docs.px4.io/master/en/simulation/jsbsim.html#running-jsbsim-with-ros

**To Reproduce**
Steps to reproduce the behavior:
1. Install the usual Development Environment on Ubuntu LTS / Debian Linux.
1. Install a JSBSim release from the release page (opens new window):
    `dpkg -i JSBSim-devel_1.1.0.dev1-<release-number>.bionic.amd64.deb`
1. Run `make px4_sitl jsbsim_quadrotor_x`

**Expected behavior**
PX4 is started with JSBSim connected and can receive commander operations.

**Log Files and Screenshots**
After outputting vehicle parameters, PX4 outputs:
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
PX4 SIM HOST: 192.168.80.1
INFO  [simulator] Simulator using TCP on remote host 192.168.80.1 port 4560
WARN  [simulator] Please ensure port 4560 is not blocked by a firewall.
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.: PX4 SITL
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

Jaeyoung-Lim, 2021-08-01:
@zimmy87 This means that jsbsim is not running(or cannot talk to the firmware).
- Could you share the full log of the terminal
> Install the usual Development Environment on Ubuntu LTS / Debian Linux.
- Could you also share which version of the environment you are running

zimmy87, 2021-09-20:
This is happening inside Ubuntu 20.04.1 LTS running inside of WSL. 

Here's the terminal output from PX4:

```
[0/4] Performing build step for 'jsbsim_bridge'
ninja: no work to do.
[3/4] cd /home/v-dazi/PX4/PX4-Autopilot-v1.12.0/build/px4_...me/v-dazi/PX4/PX4-Autopilot-v1.12.0/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/v-dazi/PX4/PX4-Autopilot-v1.12.0/build/px4_sitl_default/bin/px4
debugger: none
program: jsbsim
model: quadrotor_x
world: LSZH
src_path: /home/v-dazi/PX4/PX4-Autopilot-v1.12.0
build_path: /home/v-dazi/PX4/PX4-Autopilot-v1.12.0/build/px4_sitl_default
SITL COMMAND: "/home/v-dazi/PX4/PX4-Autopilot-v1.12.0/build/px4_sitl_default/bin/px4" "/home/v-dazi/PX4/PX4-Autopilot-v1.12.0/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/v-dazi/PX4/PX4-Autopilot-v1.12.0"/test_data
INFO  [px4] Creating symlink /home/v-dazi/PX4/PX4-Autopilot-v1.12.0/build/px4_sitl_default/etc -> /home/v-dazi/PX4/PX4-Autopilot-v1.12.0/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=3010

     JSBSim Flight Dynamics Model v1.2.0.dev1 [GitHub build 596/commit 08d2f857b407b5501fe8bba5c245c1051c606c60] Jul 25 2021 22:51:08
            [JSBSim-ML v2.0]

JSBSim startup beginning ...

INFO  [param] selected parameter default file eeprom/parameters_3010
[param] parameter file not found, creating eeprom/parameters_3010
  SYS_AUTOCONFIG: curr: 0 -> new: 1

  Output data set: 0
  SYS_AUTOSTART: curr: 0 -> new: 3010
Reading Aircraft Configuration File: dflyer
                            Version: 2.0

This aircraft model is an ALPHA release!!!

This aircraft model may not even properly load, and probably will not fly as expected.

Use this model for development purposes ONLY!!!

  Description:   Quadrotor X
  Model Author:  Matt Vacanti
  Creation Date: 2020-01-04
  Version:       0.0

  Aircraft Metrics:
    WingArea: 0.182986
    WingSpan: 0.426509
    Incidence: 0
    Chord: 0
    H. Tail Area: 0
    H. Tail Arm: 0
    V. Tail Area: 0
    V. Tail Arm: 0
    Eyepoint (x, y, z): 0 , 0 , 13.5827
    Ref Pt (x, y, z): 0 , 0 , 13.5827
    Visual Ref Pt (x, y, z): 0 , 0 , 13.5827

  Mass and Balance:
[... 395 lines cut ...]

  Mass Properties Report (English units: lbf, in, slug-ft^2)
                                      Weight    CG-X    CG-Y    CG-Z         Ixx         Iyy         Izz         Ixy         Ixz         Iyz
    Base Vehicle                          12.4     0.0     0.0    13.6         0.7         0.7         1.5        -0.0         0.0        -0.0
0   Payload                                0.0     0.0     0.0     0.3         0.0         0.0         0.0        -0.0         0.0        -0.0
                                                                                                                        
    Total:                              12.4     0.0     0.0    13.6         0.7         0.7         1.5         0.0         0.0         0.0

End of vehicle configuration loading.
-------------------------------------------------------------------------------
  CAL_MAG1_ID: curr: 0 -> new: 197644
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
* SYS_AUTOCONFIG: curr: 1 -> new: 0
  IMU_INTEG_RATE: curr: 200 -> new: 250
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
PX4 SIM HOST: 172.23.48.1
INFO  [simulator] Simulator using TCP on remote host 172.23.48.1 port 4560
WARN  [simulator] Please ensure port 4560 is not blocked by a firewall.
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
```

I'm running this inside of a PX4 repository synced to the v1.12.3 tag

And I have v1.2.0.dev1 of JSBSim installed

Jaeyoung-Lim, 2021-09-20:
@zimmy87 It seems like jsbsim is starting, but px4 is not able to communicate with it.

jsbsim with WSL has not been supported, nor it is a tested environment. Therefore, I would say the problem is related to the wsl environment you have.

zimmy87, 2021-09-20:
Thanks for the quick response @Jaeyoung-Lim, I'll pass that info along to my team and continue testing in a full installation of Ubuntu