PX4/PX4-Autopilot #17995: PX4 Gazebo Unable to Render Textures

TomoCoder123, 2021-07-30:
I am running Gazebo alongside PX4 and in the custom world I created, it seems that the textures are not loading in for the trees (pine tree model). It seems to use images from my screen to create the textures as sometimes Gazebo Menus can be shown on the tree textures in place of the white areas. The white poles on the left are actually trees and they should look more like the trees on the right. Also the environment in the back should just be the VRC driving course from the gazebo models but that is also not rendered.
[image]
Thanks in advance

mengchaoheng, 2021-08-10:
good job， maybe a gazebo problem？

Jaeyoung-Lim, 2021-08-20:
Closing as a duplicate of https://github.com/PX4/PX4-SITL_gazebo/issues/789

closed 2021-08-20 by Jaeyoung-Lim