PX4/PX4-Autopilot #18014: Hardfault on FMUK66

dk7xe, 2021-08-02:
**Describe the bug**
I'm periodically getting a hardfault on FMUK66

**To Reproduce**
Steps to reproduce the behavior:
1. compile current master for target nxp_fmuk66-v3_socketcan
2. update FMUK66_v3 with fmuk66-v3_socketcan build
3. Power cycle FMU
4. See error

**Expected behavior**
Normal operation

**Log Files and Screenshots**
[image]
full hardfaul message:
[FMUK66_hardfault.txt](https://github.com/PX4/PX4-Autopilot/files/6916049/FMUK66_hardfault.txt)

```
nsh> ver all
HW arch: NXP_FMUK66_V3
FW git-hash: 607be59fd5b47160326052dea8eea7edc7619d66
FW version: 1.12.0 0 (17563648)
FW git-branch: master
OS: NuttX
OS version: Release 10.0.0 (167772415)
OS git-hash: eeaa803749e44e1b233cf35964f588f57777833c
Build datetime: Aug  2 2021 10:06:00
Build uri: localhost
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 0003000dffffffffffff4e4555526005002c
MCU: Kinetis K66, rev. 8
nsh>
```

dk7xe, 2021-08-02:
@dagar @davids5

davids5, 2021-08-03:
@dk7xe  would you please test https://github.com/PX4/PX4-Autopilot/pull/18017

It fixes a bad stop but there is still massive leaking.

Also, please list what HW is connected and command you have issued

davids5, 2021-08-03:
@dk7xe 

 hmmm.

do you have code on your branch?

I can not check out that hash:
```
` git checkout 607be59fd5b47160326052dea8eea7edc7619d66
fatal: reference is not a tree: 607be59fd5b47160326052dea8eea7edc7619d66`
```

dk7xe, 2021-08-09:
Hi @davids5 
since #18017 was merged to master i tested a fresh master built.
Now the hardfault comes when i arm the rover during that try.
[FMUK66_hardfault_202108091055.txt](https://github.com/PX4/PX4-Autopilot/files/6953136/FMUK66_hardfault_202108091055.txt)

Afterwards the hardfault came at the moment of home position set.

davids5, 2021-08-10:
Hi @Gerald I am going to need some preproduces help on the HF.
```
HW arch: NXP_FMUK66_V3
FW git-hash: e9a21ee9110d3d25b1d330c3967c79d409da51a3
FW version: 1.13.0 40 (17629248)
FW git-branch: master
```
Is not PX4 master. Can you check your remote

This is PX4
```
FW git-hash: 11480cdbc7cee0ee0c05098555a912d1dfa5d357
FW version: 1.12.0 0 (17563648)
FW git-branch: master
OS: NuttX
OS version: 0.0.0 0 (0)
OS git-hash: eeaa803749e44e1b233cf35964f588f57777833c
Build datetime: Aug  9 2021 13:20:59
```
Maybe we can work thought it on a call. I am going to need the params set and commands that are causing the fault.

closed 2023-11-01 by dk7xe