PX4/PX4-Autopilot #18016: Failure injection for system components does not work

aksuns96, 2021-08-03:
**Describe the bug**
Failure injection for system components (motor, servo, battery, etc.) returns "Timeout waiting for ack" and fails on the release branch.

**To Reproduce**
Steps to reproduce the behavior:
1. Build and run px4_sitl_default on the release branch (9e4a04f58ab655c5129c0349c69cf70f472a3c51)
2. Set SYS_FAILURE_EN parameter to 1 
3. Takeoff
3. From the system console execute "failure motor off" 
4. See the error

**Expected behavior**
Motors should stop working when the failure command executes.

**Screenshot**
[image]

KimHyungSub, 2021-08-04:
I also noticed that failure injection feature is broken.
When "failure gyro off" or "failure accel off" are executed on the command window, PX4 keeps returning "ERROR [simulator] poll timeout 0, 22" errors, and connection between PX4 and QGC is disconnected.

[image]

aksuns96, 2021-08-04:
I think airspeed failure is broken as well. "failure airspeed off" works as intended but if you use "failure airspeed ok", the airspeed reading on QGC stays at 0 as if it is still off.

jeremyyano, 2021-11-30:
I'm also reproducing this error while testing rc_signal failure.

[image]

Using px4_sitl jmavsim in Ubuntu 20 LTS VM.

zlowell2, 2022-03-30:
Has there been any status update on this? All system failure injections seem to fail, but component injections work just fine.

castellucci5, 2022-04-07:
Hello, is there any news about it?
A few clues on what to modify in order to use "failure motor off"? 
Thank you all

joshuasv, 2024-05-29:
Same problem here

[image]

[image]

bresch, 2024-05-30:
Relates to https://github.com/PX4/PX4-Autopilot/issues/22296 .

@julianoes Could you comment on this? I think there is a misunderstanding on how is supported and on which simulator. Maybe the "failed" error message should be a bit more descriptive. e.g.: "Not implemented"

julianoes, 2024-05-30:
I assume it's just broken and needs fixing. I'd have to dive into the code and check what's going on. Sorry, I know that's not very helpful.

serkanMzlm, 2024-07-17:
You can create a fault using the commands specified on [this](https://docs.px4.io/main/en/modules/modules_system.html#sensor-gps-sim) page.

example: 
```
sensor_gps_sim stop
```

aaronmakerain, 2025-05-17:
Has anyone been able to look into this issue? I'm seeing the same "Timeout waiting for ack" issue when running failure commands.

flyingpafnouti, 2025-06-22:
Experimenting the same "Timeout waiting for ack" with px4_sitl x500_gz. I have  PX4 version: 1.16.0 c0 (17825984). 
I want to test my  visual odometry node and can't arm without GPS... I don't even speak of offboard...

david-guardian, 2025-07-29:
review 1.16, it looks like this is not yet implemented for newer simulators. It looks to me like the only failure mode implemented for modern Gazebo is the motor failure?
https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/commander/failure_detector/FailureDetector.cpp#L61

`sensor_gps_sim stop` on 1.16 does nothing as this isn't actually running by default apparently. I can start it but that adds a second gps, which I then can't stop either.

julianoes, 2025-10-31:
The issue with this one is that if you disable motor outputs with SITL, lockstep gets out of sync and everything stalls.

If you want to simulate this, you have to disable lockstep: https://docs.px4.io/main/en/sim_gazebo_classic/#disabling-lockstep

closed 2025-10-31 by julianoes