PX4/PX4-Autopilot #18018: PX4 as a static library

eden-desta, 2021-08-03:
Hi! 
I have been playing around with this lib and it is phenomenal! My main question stems from wanting to use only certain components of PX4 (for instance the uORB communcation -- especially since i was able to see it working with ROS2 which was awesome). 

I already have a NuttX application. For the purpose of simplicity lets assume it is as basic as 'Hello World' example application. I want to essentially pull in the different components I need such as uORB by making it a static library that i can just access. 

Is this something that has been done or you have seen people do? How can i go about being able to do this!

Thanks in advance! Look forward to your response!

dagar, 2021-08-03:
You could pull these things apart, but it will be kind of awkward from a built system perspective.

My recommendation would be to do it as a stripped down "PX4 build" will all the modules and drivers removed. Each module is effectively the same as a NutX app, but with full access to PX4 uORB, parameters, etc.

For example here's a full flight controller (px4_fmu-v5_default). https://github.com/PX4/PX4-Autopilot/blob/master/boards/px4/fmu-v5/default.cmake

Here's a much smaller CAN sensor node build.
https://github.com/PX4/PX4-Autopilot/blob/master/boards/ark/can-flow/default.cmake

eden-desta, 2021-08-04:
@dagar thank you for your response!
Can you elaborate on what you mean when you say pull them apart. And okay I see the stripped down version and the full flight controller. Are the things like uorb automatically built in?

Sorry I am completely new to px4 so I have a million questions. So to elaborate. You mention that each module and driver is essentially the Same as a nuttx app but with full access to px4 features (ie probably why you don't need to mention uorb in the file). 

If I remove all of the drivers and modules and make a cmake specific for my application, how do I go about accessing these features?

dagar, 2021-08-04:
> Can you elaborate on what you mean when you say pull them apart.

It would be a pain to try and build and use only uORB in another project, but it would be pretty easy to do a minimal PX4 build that's effectively nothing but NuttX + PX4 uORB. 
 

> and i notice that uorb is part of the SYSTEMCMDS.

That's for command line interactions like `uorb top`, within PX4 you always get uORB pub/sub internally from the platform layer.

> Do you essentially mean I should make a board cmake with only uorb in the systemcmds if for instance that is all i want to use

Yes!

> Additionally, if I remove all of the drivers and modules and make a cmake specific for my application, how do I go about accessing these features?

Which features? You can add or remove whichever subset of the PX4 drivers/modules you want. Then within PX4 is a complete NuttX build where you can entirely work at that level if you prefer.

For example with the common PX4 board px4_fmu-v5 (boards/px4/fmu-v5) I can run `make px4_fmu-v5_default menuconfig` and it runs the regular NuttX menuconfig. PX4 is a superset of the NuttX build.