PX4/PX4-Autopilot #18025: Precision landing using external computer vision

blakermchale, 2021-08-05:
**Describe problem solved by the proposed feature**
This will allow for precision land to be used with external computer vision algorithms. Examples would be landing on ArUco markers or a detected object. Currently, only an IR beacon is supported or documented.

**Describe your preferred solution**
When precision land mode is activated it will wait to receive poses of the object to land on over mavlink. Using these poses it will navigate to that location and land. Ardupilot seems to have an implementation already and examples can be seen in this [discussion](https://discuss.ardupilot.org/t/vision-based-precision-landing/52670).

**Describe possible alternatives**
Not sure.

**Additional context**
N/A

potaito, 2021-08-05:
> Currently, only an IR beacon is supported or documented.

Could be that the docs are confusing, but this is not true. All you need to do is publish the [LANDING_TARGET](https://mavlink.io/en/messages/common.html#LANDING_TARGET) message to the drone. Inside PX4 this is then translates to the [irlock](https://github.com/PX4/PX4-Autopilot/blob/master/msg/irlock_report.msg) message, whose naming is unfortunate because it works with any kind of target, not just IR beacons of course.

So all you need to do is implement your CV pipeline offboard and then publish the LANDING_TARGET mavlink message. Then configure the drone to perform precision landing as described in the docs.

blakermchale, 2021-08-05:
Thanks! I had tried that using RTPS though I wasn't able to get it working. Does this require switching to precision land mode? Also, would you know of any docs detailing the steps to get this working? I had asked about this in the slack [here](https://app.slack.com/client/T0V5F863B/C581ST17V/thread/C581ST17V-1628115657.009600) and also found this [discussion](https://discuss.px4.io/t/landing-target-estimator-module-for-non-irlock/18478) though there didn't seem to be any direct solutions.

TSC21, 2021-08-05:
> Btw the px4 message I was publishing with RTPS was [`landing_target_pose`](https://github.com/PX4/PX4-Autopilot/blob/master/msg/landing_target_pose.msg). Not sure if this corresponds to the `LANDING_TARGET` mavlink message.

It does and you can perfectly use it. Just note that you need to send the absolute position of the target and be on precision landing mode for it to work. You can mimic the expected configuration from https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L2393-L2402.

potaito, 2021-08-05:
Exactly! Either you publish one-off measurements via `LANDING_TARGET`, which then run through the onboard **landing_target_estimator** estimator in px4 and produce `landing_target_pose`, or you have your own estimator and publish the estimate `landing_target_pose` yourself. 

In that case you only need to change one parameter in navigator 
```
/**
 * RTL precision land mode
 *
 * Use precision landing when doing an RTL landing phase.
 *
 * @value 0 No precision landing
 * @value 1 Opportunistic precision landing
 * @value 2 Required precision landing
 * @group Return To Land
 */
PARAM_DEFINE_INT32(RTL_PLD_MD, 0);
```

and navigator should then consume your `landing_target_pose` message and land accordingly.

blakermchale, 2021-08-05:
Thanks, I got it working by following those steps!

closed 2021-08-05 by blakermchale

potaito, 2021-08-06:
Awesome!

REGATTE, 2022-02-25:
> Thanks, I got it working by following those steps!

Hi, 
could you please provide a link to the repository with the work? I would like to understand the process of action in detail.

potaito, 2022-02-25:
@REGATTE I think he was referring to these steps: https://github.com/PX4/PX4-Autopilot/issues/18025#issuecomment-893235601

REGATTE, 2022-02-25:
@potaito I did see that. I was wondering if @blakermchale has built a working Precision Landing System with ROS2 so that I could infer from that.

blakermchale, 2022-02-25:
@REGATTE I didn't end up using the Precision Landing System with ROS2 since it didn't work exactly how I wanted so I don't have any examples