PX4/PX4-Autopilot #18029: FMUv5X incorrect rotation when calibrating

mrpollo, 2021-08-05:
When calibrating Holybro FMUv5X with a Holybro GPS using v1.12, the GPS reports incorrect mag #17765, this issue was initially reported by Farhang.

[image]
[image]

https://user-images.githubusercontent.com/317648/128392759-f6c12f62-91cf-4d9b-acc6-e57cdc125ecf.mp4

mrpollo, 2021-08-11:
cc @davids5

dagar, 2021-08-11:
https://github.com/PX4/PX4-Autopilot/blob/a6d88cad18a52c39c9baf46e7a4ebf953e0293a3/boards/px4/fmu-v5/init/rc.board_sensors#L24-L25

davids5, 2021-09-17:
fixed on master by https://github.com/PX4/PX4-Autopilot/pull/18069

closed 2021-09-17 by davids5