PX4/PX4-Autopilot #18034: When EKF2 is fused with Vison position, the local x and y positions will diverge

wangrui996, 2021-08-08:
I use TX2 to run the visual inertial odometer as the positioning source, and send the positioning data to px4 through the topic "/mavros/vision_pose/pose" for fusion.
The fusion of data seems to be successful, because I compared the two topics of "/mavros/vision_pose/pose" and "mavros/local_position/pose", they are basically the same, and the changes when moving in a small area are also correct.
So I tested the drone in the fixed Altitude mode.The reason why I was not in the Position mode was because I wanted to adjust the pid parameter first, But I think this shouldn't be the point, right?
After the flight test, the log file was analyzed and found that the local x and y coordinates after ekf2 fusion diverged, but the z coordinates looked normal.
Set Up:
SYS_MC_EST_GROUP = ekf2
EKF2_AID_MASK:24(vision position fusion and vision yaw fusion)
EKF2_HGT_MODE: Vision
Pixhawk model is 2.4.8
The firmware version is PX4 1.9.2.
The z-axis vison positioning data looks basically the same as the px4 data
This is the z axis
<p align="center">[image]</p>
This is the x-axis, the yellow represents the visual data, and the blue is the px4 after ekf2, and the y direction is similar.
<p align="center">[image]</p>
What I can be sure is that there is no problem with my visual positioning data, there is no divergence, and the data is correct. PX4 has also successfully performed ekf2 fusion. My flight control used vibration reduction. So I don't understand where the problem is. 
For this reason, I just held the drone for a mobile test while observing the visual trajectory and the estimated trajectory after fusion of px4. Even if the drone does not have high frequency vibration and moves quickly and greatly, the estimated x and y coordinates of px4 will be the same, suddenly deviated a lot, up to 10 meters. and it can be determined that the visual trajectory is correct.Although they will be pulled back in a short period of time, the fusion data of x, y and the visual positioning data will not be able to continue to fusion well. The phenomenon in the log file during the actual flight is the same..  So I guess it may be a problem with EKF2, but I am not an expert in this area. I don't know if my related parameters are not set correctly or for other reasons.Is there a teacher who can help me analyze it together!

I would be very grateful if you can get help from your message!

xdwgood, 2021-08-09:
I have two suggestions: 
1: Update to v1.12 
2: Provide a completed flight log instead of partial screenshots :smiley:

wangrui996, 2021-08-14:
@xdwgood Hi，Thank you for your suggestion, and I am very sorry to reply you so late.I will further test after preparation