PX4/PX4-Autopilot #18043: How to shutdown Heater Element in 1.12.1 ?

darthuh, 2021-08-10:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or Slack (you can find an invite link on this project README).

**Describe problem solved by the proposed feature**
A clear and concise description of the problem, if any, this feature will solve. E.g. I'm always frustrated when ...

**Describe your preferred solution**
For my work, wanna simple fly with AP and not at all use heater for precise controller of sensors or so on.
Because I bought multi orange cubes and wanna  not warm!.

**Describe possible alternatives**
I changed some codes of 1.12.1 but the heater is hot (52 c).

**Additional context**
Can with any parameters or any other version or modify of heater.h for shutdown it?!
Thx, all.

mcsauder, 2021-08-11:
Hi @darthuh , can you please verify that SENS_EN_THERMAL is set to 0 or -1?  (This should be done by default, https://github.com/PX4/PX4-Autopilot/blob/a93f8dade4e65848cd33cbb9dd0eb10ccbbf28d1/boards/cubepilot/cubeorange/init/rc.board_defaults#L13, but if for some reason it has been set to turn on it would remain that way with your hardware until you change it back.)

I have a bench unit I will try to do some debugging with toward the weekend.  Hopefully it is just a param change for you.

mcsauder, 2021-08-11:
Hi again @darthuh , if might be an issue with your hardware as well, by default the heater driver should not even be built into the cube orange firmware: https://github.com/PX4/PX4-Autopilot/blob/a93f8dade4e65848cd33cbb9dd0eb10ccbbf28d1/boards/cubepilot/cubeorange/default.cmake#L28

dagar, 2021-08-11:
Here's what it looks like on the test rack running 24/7.

[image]

mcsauder, 2021-08-11:
Hehe, that's certainly a floating GPIO.  @AlexKlimaj , would you be willing to look at this for a moment?

darthuh, 2021-08-11:
> SENS_EN_THERMAL

thx, but in QGC this parameter set in  "Thermal control off" and mentioned as default: -1

darthuh, 2021-08-11:
As mentioned this [with](https://github.com/PX4/PX4-Autopilot/issues/11832) with @jzazbert ardupilot firmware all lights are off but after few seconds this board is warm.

mcsauder, 2021-08-11:
Sorry to ask just to clarify, but, is it the same issue with ardupilot firmware, or not the same case?

mcsauder, 2021-08-11:
Actually, regardless, this still feels like a floating gpio; not something being driven high by software. Any thoughts are welcome.

darthuh, 2021-08-11:
> Sorry to ask just to clarify, but, is it the same issue with ardupilot firmware, or not the same case?

I say this cause of  @jzazbert  comment. In Ardu has some tempretures in Analysis page (QGC) for average is 52 C.
I wanna to set it below 40

Saleh-At, 2021-08-15:
> What is the temperature on 1.11?

It does not different

dagar, 2021-08-15:
At a glance I don't see any problem with the heater configuration. On other boards I've noticed the STM32H7 is a bit hotter than the typical F4 or F7 we've used before. If you really want to reduce the temperature you could consider lowering the CPU clock, the H7 is currently excessively fast for how we're using it.

Saleh-At, 2021-08-20:
Is there step by step tutorial?

closed 2025-11-30 by github-actions[bot]