PX4/PX4-Autopilot #18044: HereFlow module with Cube and Mini Carrier Board

diablo1281, 2021-08-10:
Hi, I encourage problem with connecting HereFlow module with Cube using mini carrier board. This board has only CAN2 connector and with this bus HereFlow doesn't work. I checked it with full size carrier board on both CAN1 and CAN2 - HereFlow work only then connected to CAN1.

Is there any quick solution to this?

dagar, 2021-08-10:
Which Cube is this exactly?

What does `uavcan status` show on the mini carrier board?

diablo1281, 2021-08-11:
I try it with both CubeBlack and CubeOrange. With sensor connected to CAN1 `uavcan status` look like this:
[image]
[image]

Sensor connected to CAN2:
[image]
[image]

taileron, 2021-09-24:
Exactly the same behaviour occurs at MRO Ctrl Zero H7 OEM.
Only at UavCan 1 the HereFlow gets recognised.

samostopar, 2023-12-12:
Hi, I have the same problem, with CUBE and Mini carrier board...