PX4/PX4-Autopilot #18048: Add Support for ARK Flow and ARK GPS Software CAN Termination

AlexKlimaj, 2021-08-10:
The ARK GPS and ARK Flow both have a circuit to enable a 120 ohm CAN termination through firmware. It is connected to PB13 on both boards. Active high to enable to termination. 

I'd like to add a feature to automatically turn this on if the cannode is the only device on the bus. For multiple devices on the bus, it should be a parameter on the cannode that can be turned. The system integrator would be able to turn it on for the last device on the bus.

https://github.com/ARK-Electronics/ARK_Flow/blob/main/Hardware/Rev%202/ARK%20Flow%20Rev%202%20Schematic.PDF
https://github.com/ARK-Electronics/ARK_GPS/blob/main/Hardware/Rev%202/ARK%20GPS%20Rev%202%20Schematic.PDF

[image]

For short CAN harnesses, you can usually get away without a termination, although is not technically correct. For long wire harnesses, the termination is critical for optimal performance. Currently, users must plug in a 120 ohm termination to a connector on the last device on the bus. See example below for external hardware solution.

https://shop.zubax.com/collections/cables-connectors/products/uavcan-micro-termination-plug?variant=6007985111069

dagar, 2021-08-10:
How sketchy would it be to set automatically? 

https://github.com/PX4/libuavcan/blob/29e1aa6c12e3590252b102153d8c62155047a676/libuavcan/src/transport/uc_can_io.cpp#L331-L334

AlexKlimaj, 2021-08-10:
I don't think there is risk in setting it automatically when there is only one peripheral on the bus. Risky if there are multiple. I don't know how software can determine where peripherals are on the bus.

rworrall, 2021-08-25:
We are using a termination resistor at the end of the 'bus'.  Our plans include multiple UAVCAN devices which do share the 1Mbps link and termination is indicated.  I'll think on this one, good discussion.

closed 2021-12-01 by AlexKlimaj