PX4/PX4-Autopilot #18049: No local position is published when using arkflow

eldewany, 2021-08-10:
**Describe the bug**
When using arkflow on CAN1 and setting EKF2_AID_MASK , no local_position message is sent neither in QGC or in mavros which is weird as it's supposed to fuse the reading with it also the height estimation is not sourced from the range sensor as set in the parameter EKF2_HGT_MODE. Also the QGC always shows extra parameters selected in the aid mask even if I unchecked them and restarted.

**To Reproduce**
Steps to reproduce the behavior:
On FW 1.12.1 connect the arkflow and set EKF2_AID_MASK to 1 and check the mavlink inspector, no local_position messages will show

**Expected behavior**
Position should be fused and altitude should be correctly affected by arkflow changes with high precision.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
[image]

[image]

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

dagar, 2021-08-10:
That's a weird QGC bug I need to look into. For now can you try setting the parameter manually (advanced or command line)?

``` Console
nsh> param set EKF2_AID_MASK 3
```

If there's still an issue after that can you share the flight log? https://logs.px4.io

eldewany, 2021-08-10:
> That's a weird QGC bug I need to look into. For now can you try setting the parameter manually (advanced or command line)?
> 
> ```
> nsh> param set EKF2_AID_MASK 3
> ```
> 
> If there's still an issue after that can you share the flight log? https://logs.px4.io

I Don't think I would fly with this condition to be honest, even after setting the mask to 3 the local position appeared but reading are the same as raw pixhawk with no arkflow

AlexKlimaj, 2021-08-11:
Yeah, I've run into the EKF2_AID_MASK setting issue as well. Selecting manual entry in QGC and entering 3 works as well. Where is the flow mounted on the vehicle? How close is it to the ground before take off? If you pick up the vehicle, will it go into position mode? In the Mavlink Inspector, are you getting distance sensor and optical flow readings?

dagar, 2021-08-11:
It depends on your set up, but what I would do is takeoff and fly a bit in stabilized, if that seems alright switch to altitude, if that's holding well then switch to position. If anything seems off then immediately step back position -> altitude -> stabilized.

Alternatively you could also just fly around in stabilized briefly and from that log we'll be able to see if flow is working well or not.

eldewany, 2021-08-11:
> Yeah, I've run into the EKF2_AID_MASK setting issue as well. Selecting manual entry in QGC and entering 3 works as well. Where is the flow mounted on the vehicle? How close is it to the ground [...]

Here's my full setup and correct me if i'm wrong: PX4 v1.12.1

EKF2_AID_MASK: 3 (Manual entry)
EKF2_HGT_MODE: Range sensor ( is this correct for height estimation of the arkflow ?)

The sensor is mounted on the front with X axis aligned with that of the pixhawk ( Can connectors pointing to the back) and SENS_FLOW_ROT is set to 0

The flow is around 20 cm high from the ground when the drone is on the ground.

I haven't tried position mode but I will try and get back to you.

I'm getting optical_flow_rad messages and distance sensor messages.

eldewany, 2021-08-11:
> It depends on your set up, but what I would do is takeoff and fly a bit in stabilized, if that seems alright switch to altitude, if that's holding well then switch to position. If anything seems off [...]

okay I will give it a shot and get back to you.

dagar, 2021-08-11:
> EKF2_HGT_MODE: Range sensor ( is this correct for height estimation of the arkflow ?)

I wouldn't use this unless you're flying indoors where it's completely flat absolutely everywhere (eg warehouse). I find the better combo to be EKF2_HGT_MODE barometer (default) with EKF2_RNG_AID enabled. Then you get the benefit of range height mode when you're still or moving slowly, otherwise it's fully functional on baro and switches automatically. This also lets you fly over objects (eg a box) without shifting upwards. Best of both worlds.

eldewany, 2021-08-11:
> I wouldn't use this unless you're flying indoors where it's completely flat absolutely everywhere (eg warehouse). I find the better combo to be EKF2_HGT_MODE barometer (default) with EKF2_RNG_AID [...]

Okay I will give it a try , thank you for the support

hichamhendy, 2022-11-30:
@eldewany can you confirm if the LED is/was solid red or turning to blue? Do you still have the same problem?
I'm struggling with that in the meantime. It worked at the beginning; yet now not more. The led is solid red and mavlink inspector doesn't show the optical flow message used to have

AlexKlimaj, 2022-11-30:
> @eldewany can you confirm if the LED is/was solid red or turning to blue? Do you still have the same problem? I'm struggling with that in the meantime. It worked at the beginning; yet now not more. The led is solid red and mavlink inspector doesn't show the optical flow message used to have

@hichamhendy Is your flight controller sd card installed and working? The ARK Flow won't boot without an SD card in the flight controller. Its needed by PX4 for the cannode database.

hichamhendy, 2022-11-30:
Ja I double-checked on that `df -h` shows the sd and I can go in the directory  `cd`.

AlexKlimaj, 2022-11-30:
> Ja I double-checked on that `df -h` shows the sd and I can go in the directory `cd`.

Try formatting the SD card

hichamhendy, 2022-11-30:
I've not tried that yet. However, I used the same SD (as it is) on another flight controller (cube orange) and it worked on the version 1.13. I then downgraded to 1.12.3 and got the same behavior again as by Holybro on both versions 1.13 and 1.12.3!!
I also built the arkflow and took XX.uav.bin and placed it in the sd root directory then flashed it through `uavcan start fw` and in the folder ufw got 88.bin generated as expected, yet the error still exits.

AlexKlimaj, 2022-11-30:
@hichamhendy It should be 80.bin for the ARK Flow. See attached binary built of main.

[80-0.1.c5dc1221.uavcan.bin.zip](https://github.com/PX4/PX4-Autopilot/files/10128179/80-0.1.c5dc1221.uavcan.bin.zip)

hichamhendy, 2022-12-01:
@AlexKlimaj  I repeated the procedure and you're right it was 80. I must have forgotten.
I also formatted the SD and it worked just fine; yet, with 1.13 not 1.12! Could you imagine where the problem could be??

Many thanks for the support

AlexKlimaj, 2022-12-02:
I added support for the flow after 1.12.

hichamhendy, 2022-12-02:
Where can I find it?

AlexKlimaj, 2022-12-07:
Send me a message on discord and we can get this figured out. I'm not sure I understand your problem.

hichamhendy, 2022-12-07:
What is your username and tag so that I can find you?

AlexKlimaj, 2022-12-07:
It's just Alex Klimaj. AlexK#9105

AlexKlimaj, 2022-12-08:
It appears that the issue is with 1.12 where the CAN firmware server only runs on CAN1 and not CAN2. Resulting in the ARK Flow not booting when connected to CAN2 on 1.12. This was fixed in this commit.

https://github.com/PX4/PX4-Autopilot/commit/2e2ac36cab867fae38e2180edbc3159672e9020d