PX4/PX4-Autopilot #18060: UAVCAN airspeed sensor cannot be calibrated

joibgbg123, 2021-08-12:
**Describe the bug**
After the v1.12.1 release the airspeed sensor cannot be calibrated if using UAVCAN sensors. I have 4 pcs pixhawk4 mini's with GPS from Holybro. One has an analog airspeed sensor from Holybro. One has the digital version connected via UAVCAN : https://docs.avionicsanonymous.com/devices/microadc#configuration. 

A few days ago I decided that all the 4 would need firmware upgrade, resetting parameters to have them all with the same parameters. This worked fine until I connected this digital UAVCAN airspeed sensor, now I couldn't calibrate the sensor. I simply got the message and error „No airspeed sensor found“.

After reviewing all parameters over and over again, and updating the firmware once again, believing that this was a problem on my end. A fellow college suggested that I tried the firmware version that I had before. I installed stable release v1.11.3 and I could do this without any problems. 

I would like to add that I could read the airspeed sensor, just not calibrate it.

**To Reproduce**
Steps to reproduce the behavior:
1. Pixhawk4 mini fw upgrade to v1.12.1 (i tried px4_fmu-v5_fixedwing.px4) and px4_fmu-v5_default.px4)
2. Have an airspeed sensor connected via UAVCAN
3. Press calibrate and a moment later it says: "No airspeed sensor found". see screenshot.

[image]

I would like to add if not clear, that nothing more is connected to the PX4 mini, I'm using those computers for measuring using the sensors and this seemed the best sensor with respect to price. 

Hope this is clear, 
best regards
Jóhannes

dagar, 2021-08-14:
Thanks for reporting, I actually didn't realize any UAVCAN airspeed sensors were readily available for purchase. 

Let me see if I can fill in what's missing.

dagar, 2021-08-14:
Can you capture the output of `uavcan status`?

dagar, 2021-08-14:
@joibgbg123 are you getting airspeed directly from UAVCAN or differential pressure?

In addition to `uavcan status` can you grab `listener airspeed` and `listener differential_pressure`?

joibgbg123, 2021-08-14:
I will acquire those command-screenshots when I get back to work on Monday morning (GMT). 
The airspeed sensor is giving differential pressure, since the PX4 is reported to like better to calculate the airspeed given the correct differential pressure. 

I actually used the command status while debugging myself, I forgot to include a screenshot of that. This command showed that the UAVCAN is talking the a differential pressure sensor, so I had a node id for a differential pressure, and no info for airspeed(Since the Pixhawk thinks none exists). The parameters are sent to the sensor to publish only raw pressure difference apposed to fully calculated airspeed. 
And like I mentioned in the original post, I can read values in the "Mavlink Inspector" tab for airspeed. Which is calculated onboard the Pixhawk from read values of differential pressure acquired from the sensor.

joibgbg123, 2021-08-17:
Hello, sorry for the late response, the day yesterday went by faster then ever. Here is the outcome of `uavcan status`:

-------------------------------------------------------------------------------------------------------
`NuttShell (NSH) NuttX-10.0.0
nsh> uavcan status
Pool allocator status:
    Capacity hard/soft: 500/250 blocks
    Reserved:  155 blocks
    Allocated: 34 blocks

UAVCAN node status:
    Internal failures: 0
    Transfer errors:   0
    RX transfers:      2293
    TX transfers:      3746

CAN1 status:
    HW errors: 17176
    IO errors: 18373
    RX frames: 4197
    TX frames: 2991
CAN2 status:

control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
INFO  [mixer_module] Switched to rate_ctrl work queue: 0
INFO  [mixer_module] Mixer loaded: no
INFO  [mixer_module] Driver instance: 0
INFO  [mixer_module] Channel Configuration:
INFO  [mixer_module] Channel 0: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 1: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 2: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 3: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 4: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 5: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 6: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 7: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 8: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 9: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 10: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 11: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 12: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 13: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 14: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191
INFO  [mixer_module] Channel 15: value: 0, failsafe: 0, disarmed: 65535, min: 1, max: 8191

Sensor 'baro':
name: uavcan_baro
channel 0: node id 101 --> instance 1
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'mag':
name: uavcan_mag
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'gnss':
name: uavcan_gnss
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty
uavcan: gnss: rtcm pub: 0 events, 0.00us avg, min 0us max 0us 0.000us rms

Sensor 'flow':
name: uavcan_flow
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'battery':
name: uavcan_battery
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'airspeed':
name: uavcan_airspeed
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'differential_pressure':
name: uavcan_differential_pressure
channel 0: node id 101 --> instance 0
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'rangefinder':
name: uavcan_rangefinder
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'accel':
name: uavcan_accel
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Sensor 'gyro':
name: uavcan_gyro
channel 0: empty
channel 1: empty
channel 2: empty
channel 3: empty

Online nodes (Node ID, Health, Mode):
     101 OK         OPERAT    

uavcan: cycle time: 55896 events, 3194035us elapsed, 57.14us avg, min 20us max 2778us 163.653us rms
uavcan: cycle interval: 55896 events, 2657.74us avg, min 32us max 8539us 992.463us rms
nsh> `
 
-------------------------------------------------------------------------------------------------------
Here is the outcome from `Listener airspeed` and `listener differential_pressure`

-------------------------------------------------------------------------------------------------------
nsh> listener airspeed

TOPIC: airspeed
 airspeed_s
    timestamp: 344966295  (0.034844 seconds ago)
    indicated_airspeed_m_s: 0.7336
    true_airspeed_m_s: 0.7454
    air_temperature_celsius: 23.6000
    confidence: 1.0000
nsh> listener differential_pressure

TOPIC: differential_pressure
 differential_pressure_s
    timestamp: 362867103  (0.081878 seconds ago)
    error_count: 0
    differential_pressure_raw_pa: 0.4275
    differential_pressure_filtered_pa: 0.4139
    temperature: 23.6000
    device_id: 8611075 (Type: 0x83, UAVCAN:0 (0x65)) 
nsh> 

-------------------------------------------------------------------------------------------------------

Here is also a screenshot for the values I'm reading:
[image]

Where the high values were produced as I blew into the pitot.

Hope this information is good.

Best regards
Jóhannes Bergur

hhenryy, 2021-08-20:
I am also interested into solving this issue. If you have any pointers were I can start let me know.

hhenryy, 2021-08-25:
I have been looking into this, but it seems it could be solved on multiple ways. Is there any proposed method? 

First time I am diving into PX4 drivers, so could be missing an obvious way.

dagar, 2021-08-25:
I found the problem, the calibration is not being handled by uavcan differential pressure. 

This won't be a problem after https://github.com/PX4/PX4-Autopilot/pull/15853 comes in, but let me see about a short term immediate fix we can get in for v1.12.2.

dagar, 2021-08-25:
Could you give https://github.com/PX4/PX4-Autopilot/pull/18145 a try?

Neliparagraphen, 2022-04-06:
I have the same issue with the orange cube and differentialpressure sensor Mateksys ASPD-DLVR. 
Is there any short term fix for this problem yet?

closed 2022-04-26 by dagar