PX4/PX4-Autopilot #18061: EKF2 Status 1.12 until master

taileron, 2021-08-12:
**Describe the bug**
all controllers without multi ekf enabled show missed messages approx the double number of events 

**To Reproduce**
Steps to reproduce the behavior:
Take a Pixracer or Omnibus F4 or another F4 based FC
nsh> ekf2 status
ekf2:0 attitude: 1, local position: 0, global position: 0
ekf2: ECL update: 14437 events, 115552us elapsed, 8.00us avg, min 6us max 171us 17.533us rms
ekf2: ECL full update: 14483 events, 8320165us elapsed, 574.48us avg, min 238us max 1716us 806.863us rms
ekf2: IMU message missed: 28921 events
ekf2: vehicle_air_data messages missed: 0 events
ekf2: vehicle_magnetometer messages missed: 0 events

**Expected behavior**
0 IMU messages missed like it is with multi ekf

dagar, 2021-08-12:
It's a harmless bug, the perf counter wasn't being updated in the single EKF2 case.

I can backport the fix for v1.12.2.

Thanks for reporting.

taileron, 2021-08-12:
@dagar in 1.12.1 there is another harmless bug left that would good fit into a 1.12.2 backport
rtl always climbs to at least 30m even if lower values e.g. 22m are set. (it seems to be fixed in master)

dagar, 2021-08-12:
> rtl always climbs to at least 30m even if lower values e.g. 22m are set. (it seems to be fixed in master)

Is this different than master? Can you open an issue with a log showing that?

dagar, 2021-08-12:
Bad reply timing, thanks for the update. I'm surprised it's different in master.

dagar, 2021-08-12:
This is where the IMU perf count was fixed in master. https://github.com/PX4/PX4-Autopilot/commit/251f1a069b1f235738e2e2e7b43cf8a1fca05761

dagar, 2021-08-12:
Simpler fix in the v1.12 release branch (will be tagged v1.12.2 in the coming weeks). https://github.com/PX4/PX4-Autopilot/commit/c5f82ed838a6996c69e0911d527ad685767579dc

taileron, 2021-08-16:
even if it seems to have no effect in operation it is still present in Master:
(PixRacer R12): ver all
HW arch: PX4_FMU_V4
FW git-hash: c6931814facd0c0a964d9b8b65d3e874599a213a
FW version: 1.13.0 40 (17629248)
OS: NuttX
OS version: Release 10.0.0 (167772415)
OS git-hash: 8660572a3c69e0e2ca24c3cabeaaaae3372a6960
Build datetime: Aug 14 2021 19:32:57
Build uri: localhost
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 0001000000003131373130365117004f0031
MCU: STM32F42x, rev. 3
nsh> ekf2 status

ekf2:0 attitude: 1, local position: 0, global position: 0
ekf2: ECL update: 13611 events, 118146us elapsed, 8.68us avg, min 1us max 198us 10.452us rms
ekf2: ECL full update: 13530 events, 8046716us elapsed, 594.73us avg, min 237us max 2536us 236.373us rms
ekf2: IMU message missed: 27143 events
ekf2: vehicle_air_data messages missed: 0 events
ekf2: vehicle_magnetometer messages missed: 0 events

taileron, 2021-08-16:
in the case of RTL, the altitude for setting e.g. RTL_RETURN_ALT = 18 has been reduced from 30m to 20m which is more compatible for small aircraft with short flight times.
https://review.px4.io/plot_app?log=4c44dd75-41bf-4be3-aa42-7d817240717d

taileron, 2022-04-11:
definitely solved with #19466

closed 2022-04-11 by taileron