PX4/PX4-Autopilot #18063: Gazebo Gimbal: Not working

tawfiq1200, 2021-08-12:
Gazebo gimbal is not responding to vmount commands or to the QGround Controller. [This](https://docs.px4.io/master/en/advanced/gimbal_control.html) is the documentation for the gimbal control using commands. The gimbal does not rotate when the commands are executed in the terminal and it also does not rotate when I set pitch and yaw to a certain degree from the QGC.
[image] The attached screenshot shows that when I run a mission from QGround control, the roll, pitch and yaw stays 0. 

I built PX4 from [this](https://docs.px4.io/master/en/dev_setup/building_px4.html) 

From this [issue](https://github.com/PX4/PX4-Autopilot/issues/8599), it seems that it could be a plugin issue, but not sure what could go wrong.

tawfiq1200, 2021-08-12:
It seems like the gimbal controller topic is not publishing:

[Err] [JointControlWidget.cc:393] Error advertising topic [/data://world/default/model/typhoon_h480/plugin/gimbal_controller/joint_cmd]

Jaeyoung-Lim, 2021-08-13:
I have just tested the gimbal on the typhoon h480, and setting ROI through QGC is fully functional.

@tawfiq1200 Could you provide the exact commands you were trying to run, and the expected behavior of the commands?

tawfiq1200, 2021-08-13:
Yes, 
commander takeoff
vmount test yaw (30, 120, etc.)

You can see from the screenshot that the gimbal is not moving.
[image]

tawfiq1200, 2021-08-13:
@Jaeyoung-Lim The drone is armed as seen in the screenshot while the gimbal is not moving.

tawfiq1200, 2021-08-13:
ROI is indeed working with QGC but not the vmount commands.

Jaeyoung-Lim, 2021-08-13:
@tawfiq1200 Which vmount command are you running?

tawfiq1200, 2021-08-13:
@Jaeyoung-Lim 
vmount stop
vmount test yaw 30

tawfiq1200, 2021-08-13:
Since QGC is working, it is not a plugin issue so I am closing the issue though updating the PX4 doc for vmount commands will be good.

closed 2021-08-13 by tawfiq1200