PX4/PX4-Autopilot #18064: make px4_sitl jmavsim not working

vibes99, 2021-08-12:
OS: macOS Big Sur Version 11.4
I am trying to build the px4 but unable to run it with jmavsim and have been continuously getting errors even thought the GUI has started and the startup script returned successfully. 

The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
Vibeeshs-MacBook-Pro:~ vibeeshbalaji$ brew upgrade
Vibeeshs-MacBook-Pro:~ vibeeshbalaji$ make clean
make: *** No rule to make target `clean'.  Stop.
Vibeeshs-MacBook-Pro:~ vibeeshbalaji$ cd firmware
Vibeeshs-MacBook-Pro:firmware vibeeshbalaji$ make clean
/Users/vibeeshbalaji/Firmware/build/px4_sitl_default
[1/1] Cleaning all built files...
Cleaning... 1142 files.
Entering 'Tools/flightgear_bridge'
Entering 'Tools/jMAVSim'
Removing out/
Entering 'Tools/jsbsim_bridge'
Entering 'Tools/simulation-ignition'
Entering 'Tools/sitl_gazebo'
Entering 'mavlink/include/mavlink/v2.0'
Entering 'platforms/nuttx/NuttX/apps'
Entering 'platforms/nuttx/NuttX/nuttx'
Entering 'src/drivers/gps/devices'
Entering 'src/drivers/uavcan/libuavcan'
Entering 'src/drivers/uavcan_v1/legacy_data_types'
Entering 'src/drivers/uavcan_v1/libcanard'
Entering 'src/drivers/uavcan_v1/public_regulated_data_types'
Entering 'src/drivers/uavcannode_gps_demo/libcanard'
Entering 'src/drivers/uavcannode_gps_demo/public_regulated_data_types'
Entering 'src/lib/crypto/monocypher'
Entering 'src/lib/events/libevents'
Entering 'src/lib/matrix'
Entering 'src/modules/micrortps_bridge/micro-CDR'
Vibeeshs-MacBook-Pro:firmware vibeeshbalaji$ make px4_sitl jmavsim
[0/807] git submodule src/drivers/gps/devices
[6/807] git submodule mavlink/include/mavlink/v2.0
[11/807] git submodule Tools/jMAVSim
[668/807] Linking CXX static library src/lib/bezier/libbezier.a
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: src/lib/bezier/libbezier.a(BezierQuad.cpp.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: src/lib/bezier/libbezier.a(BezierQuad.cpp.o) has no symbols
[685/807] Linking CXX static library src/lib/slew_rate/libSlewRate.a
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: src/lib/slew_rate/libSlewRate.a(dummy.cpp.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: src/lib/slew_rate/libSlewRate.a(dummy.cpp.o) has no symbols
warning: /Library/Developer/CommandLineTools/usr/bin/ranlib: archive library: src/lib/slew_rate/libSlewRate.a the table of contents is empty (no object file members in the library define global symbols)
[701/807] Linking CXX static library src/drivers/camera_trigger/libdrivers__camera_trigger.a
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: src/drivers/camera_trigger/libdrivers__camera_trigger.a(pwm.cpp.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: src/drivers/camera_trigger/libdrivers__camera_trigger.a(seagull_map2.cpp.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: src/drivers/camera_trigger/libdrivers__camera_trigger.a(gpio.cpp.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: src/drivers/camera_trigger/libdrivers__camera_trigger.a(pwm.cpp.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: src/drivers/camera_trigger/libdrivers__camera_trigger.a(seagull_map2.cpp.o) has no symbols
/Library/Developer/CommandLineTools/usr/bin/ranlib: file: src/drivers/camera_trigger/libdrivers__camera_trigger.a(gpio.cpp.o) has no symbols
[806/807] cd /Users/vibeeshbalaji/Firmware/build/px4_sitl_default/tmp && /Users/vibeeshbalaji/Firmwa...jmavsim none none /Users/vibeeshbalaji/Firmware /Users/vibeeshbalaji/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /Users/vibeeshbalaji/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
world: none
src_path: /Users/vibeeshbalaji/Firmware
build_path: /Users/vibeeshbalaji/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/Users/vibeeshbalaji/Firmware/build/px4_sitl_default/bin/px4" "/Users/vibeeshbalaji/Firmware/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/Users/vibeeshbalaji/Firmware"/test_data
Buildfile: /Users/vibeeshbalaji/Firmware/Tools/jMAVSim/build.xml
Creating symlink /Users/vibeeshbalaji/Firmware/build/px4_sitl_default/etc -> /Users/vibeeshbalaji/Firmware/build/px4_si

_____  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
[... 171 lines cut ...]
INFO  [simulator] Simulator connected on TCP port 4560.
Init MAVLink
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
WARN  [logger] Too many subscriptions, failed to add: collision_constraints 0
WARN  [logger] Too many subscriptions, failed to add: obstacle_distance_fused 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_mocap_odometry 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_trajectory_waypoint 0
WARN  [logger] Too many subscriptions, failed to add: vehicle_trajectory_waypoint_desired 0
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2021-08-12/15_10_22.ulg
INFO  [logger] Opened full log file: ./log/2021-08-12/15_10_22.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [tone_alarm] home set
WARN  [mc_pos_control] invalid setpoints
INFO  [tone_alarm] notify negative

I'm not sure what to do, I've tried updating the java version to 16 but unable to work out any of the commands like takeoff and landing. Can anyone help me with this?

dagar, 2021-08-18:
Can you try an older version of java?

NicolassFelipe, 2021-08-24:
Hello there, did you get any solution? i'm stuck with the same warn.

pxh> INFO  [tone_alarm] home set
WARN  [mc_pos_control] invalid setpoints
INFO  [tone_alarm] notify negative

richtong, 2021-09-09:
Hmm. this is strange, I was able to get it to work the jdk16 with the adoptopenjdk, did you use the Oracle one. I also got a very different error that complained about the wrong API level, so I don't think it is that. This was on a Mac Big Sur 11.5 though

valxv, 2021-10-30:
Having the same problem on macOS Big Sur 11.6.1.
Was able to compile only with JDK 17, but after the start have AppContextInfo(Bug 1004) caught.

potaito, 2021-12-27:
I ran into the same issue with Arch Linux when I installed `jdk-openjdk` which currently contains version 17.0.1.u12-1. After downgrading to `jre11-openjdk`, jmavsim is launching again.

alper50, 2022-03-22:
@PSandro 
Current ı am having the same issue. ı have arch linux and dont know what to do?