PX4/PX4-Autopilot #18071: micrortps_agent is not communicating with micrortps_client

a-sheikh, 2021-08-14:
**Describe the bug**
I was working with PX4 ROS2 bridge. I had to reinstall everything in attempt to solve this error 'eprosima::fastcdr::exception::NotEnoughMemoryException' whenever i subsribed to any topic. 

I reinstalled everything needed for the correct working of PX4-ROS2 bridge. Everything builds. Micrortps_agent runs, and i can list all the topics in ros enviornment. 

But now Whenever i try to run the offboard example it does not work. micrortps_agent shows that the publisher and subsrcibers but nothing happens. If i manually try to echo sensor combined or anyother topic other than timesync there is no data.

As per https://docs.px4.io/master/en/ros/ros2_offboard_control.html, i thought maybe the yaml file is not configured for recieving messages, but _PX4-Autopilot/msg/tools/uorb_rtps_message_ids.yaml_ doesnt exist.

I have no idea what to do now, as this was working days ago. @TSC21 Can you please help?

TSC21, 2021-08-14:
Latest changes brought with https://github.com/PX4/PX4-Autopilot/pull/18052 were still not brought to px4_ros_com since Jenkins is not deploying it because of problem in the pipeline. I will only be able to do that this Monday. I ask to pelase be patient for now and sorry for any inconvenience.

a-sheikh, 2021-08-18:
@TSC21 have the px4_ros_com been updated?.... it builds but still doesnt communicate

TSC21, 2021-08-18:
Yes it was yesterday. Make sure to clean your workspace first before rebuilding.

a-sheikh, 2021-08-18:
i deleted autopilot, px4_ros_com, and px4_msg. rebuild everything. still the offboard example does not work, neither does sensor_combined ...

a-sheikh, 2021-08-18:
@TSC21 here is my system setup, 
Ubuntu : 18.04
ROS : Dashing
Fast RTPS version : 1.8.2
RTPSGEN : 1.0.4

PX4 autopilot, PX4 ros com, and PX4 msgs all have been updated to the current git commits. 
still the offboard example does not work, i can see 
```
[   micrortps_agent   ]	TrajectorySetpoint subscriber matched
[   micrortps_agent   ]	Timesync publisher matched
[   micrortps_agent   ]	OffboardControlMode subscriber matched
[   micrortps_agent   ]	VehicleCommand subscriber matched
```
but no arm command on sitl. 
and
when i try to ros2 run px4_ros_com sensor_combined_listener, i see no topic matched on micrortps_agent...
am i missing something?

closed 2021-08-19 by a-sheikh