PX4/PX4-Autopilot #18077: Can Bootloaders Not Building

AlexKlimaj, 2021-08-15:
I am unable to build can bootloaders with the new Nuttx commits.

It is the same error on all can bootloaders builds. 

make ark_can-gps_canbootloader
make ark_can-flow_canbootloader
make cuav_can-gps-v1_canbootloader

```
[0/272] git submodule platforms/nuttx/NuttX/nuttx
[1/272] git submodule platforms/nuttx/NuttX/apps
[21/272] Generating nuttx/fs/libfs.a
FAILED: NuttX/nuttx/fs/libfs.a 
cd /home/alex/PX4-Autopilot/build/ark_can-gps_canbootloader/NuttX/nuttx && find fs -type f -name *.o -delete && make -C fs --quiet --no-print-directory all TOPDIR=/home/alex/PX4-Autopilot/build/ark_can-gps_canbootloader/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
dirent/fs_readdir.c: In function 'readpseudodir':
dirent/fs_readdir.c:63:3: error: 'strncpy' destination unchanged after copying no bytes [-Werror=stringop-truncation]
   63 |   strncpy(idir->fd_dir.d_name, idir->u.pseudo.fd_next->i_name,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   64 |           NAME_MAX);
      |           ~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:79: fs_readdir.o] Error 1
[24/272] Generating nuttx/libs/libc/libc.a
ninja: build stopped: subcommand failed.
make: *** [Makefile:225: ark_can-gps_canbootloader] Error 1
```

dagar, 2021-08-16:
Toolchain version?

https://github.com/PX4/PX4-Autopilot/actions/runs/1134481909

[image]

AlexKlimaj, 2021-08-16:
I ran the ubuntu toolchain script before this as well. Looks like I'm on 9.2.1

```
alex@alex-ThinkCentre-M75q-Gen-2:~/PX4-Autopilot$ make ark_can-flow_canbootloader 
-- PX4 version: v1.13.0-alpha1-2162-g474b0188f7
-- PX4 config file: /home/alex/PX4-Autopilot/boards/ark/can-flow/canbootloader.cmake
-- PX4 config: ark_can-flow_canbootloader
-- PX4 platform: nuttx
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 9.2.1
-- The C compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /bin/arm-none-eabi-g++
-- Check for working CXX compiler: /bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /bin/arm-none-eabi-gcc
-- Check for working C compiler: /bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/alex/PX4-Autopilot/build/ark_can-flow_canbootloader
[0/272] git submodule platforms/nuttx/NuttX/nuttx
[2/272] git submodule platforms/nuttx/NuttX/apps
[21/272] Generating nuttx/fs/libfs.a
FAILED: NuttX/nuttx/fs/libfs.a 
cd /home/alex/PX4-Autopilot/build/ark_can-flow_canbootloader/NuttX/nuttx && find fs -type f -name *.o -delete && make -C fs --quiet --no-print-directory all TOPDIR=/home/alex/PX4-Autopilot/build/ark_can-flow_canbootloader/NuttX/nuttx KERNEL=y EXTRAFLAGS=-D__KERNEL__
dirent/fs_readdir.c: In function 'readpseudodir':
dirent/fs_readdir.c:63:3: error: 'strncpy' destination unchanged after copying no bytes [-Werror=stringop-truncation]
   63 |   strncpy(idir->fd_dir.d_name, idir->u.pseudo.fd_next->i_name,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   64 |           NAME_MAX);
      |           ~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [Makefile:79: fs_readdir.o] Error 1
[24/272] Generating nuttx/libs/libc/libc.a
ninja: build stopped: subcommand failed.
make: *** [Makefile:225: ark_can-flow_canbootloader] Error 1
alex@alex-ThinkCentre-M75q-Gen-2:~/PX4-Autopilot$ 

```

AlexKlimaj, 2021-09-30:
I am still having this problem. I need to comment out lines 63 and 64 in dirent/fs_readdir.c to get it to build.

dagar, 2021-10-01:
I would either fix `dirent/fs_readdir.c` directly or add -Wno-stringop-truncation to Make.defs.in.

https://github.com/PX4/PX4-Autopilot/blob/68a13326f07bd8d65973658528fd63bf885ebdc1/platforms/nuttx/NuttX/Make.defs.in#L96

davids5, 2021-10-01:
Why not use the release we use?
```
RUN mkdir -p /opt/gcc && cd /opt/gcc \
	&& wget -qO- "https://developer.arm.com/-/media/Files/downloads/gnu-rm/9-2020q2/gcc-arm-none-eabi-9-2020-q2-update-x86_64-linux.tar.bz2" | tar jx --strip 1 \
	&& rm -rf /opt/gcc/share/doc
```

dagar, 2021-10-01:
> Why not use the release we use?
> 
> ```
> RUN mkdir -p /opt/gcc && cd /opt/gcc \
> 	&& wget -qO- "https://developer.arm.com/-/media/Files/downloads/gnu-rm/9-2020q2/gcc-arm-none-eabi-9-2020-q2-update-x86_64-linux.tar.bz2" | tar jx --strip 1 \
> 	&& rm -rf /opt/gcc/share/doc
> ```

Something else on his system installed a slightly older version (9.2.1) in /bin which takes precedence in his path. I'm sure we could track down the particular problem for Alex, but I'd rather have a little leeway to minimize pain for any users we won't hear from directly.

closed 2021-12-01 by AlexKlimaj