PX4/PX4-Autopilot #18092: Can't get Jinja2, toml, or empy to install at all

HatedFate, 2021-08-17:
[2/796] Generating mixer_multirotor.generated.h
FAILED: src/lib/mixer/MultirotorMixer/mixer_multirotor.generated.h 
cd /home/kannachan/px4/PX4-Autopilot/build/px4_sitl_default/src/lib/mixer/MultirotorMixer && /usr/bin/python3 /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/tools/px_generate_mixers.py -f /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/dodeca_bottom_cox.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/dodeca_top_cox.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_cox.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_plus.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_t.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_x.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_cox.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_cox_wide.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_plus.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_x.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_deadcat.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_h.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_plus.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_s250aq.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_vtail.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_wide.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x_cw.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x_pusher.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_y.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/tri_y.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/twin_engine.toml -o mixer_multirotor.generated.h
Failed to import toml: No module named 'toml'

You may need to install it using:
    pip3 install --user toml

[3/796] Generating mixer_multirotor_normalized.generated.h
FAILED: src/lib/mixer/MultirotorMixer/mixer_multirotor_normalized.generated.h 
cd /home/kannachan/px4/PX4-Autopilot/build/px4_sitl_default/src/lib/mixer/MultirotorMixer && /usr/bin/python3 /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/tools/px_generate_mixers.py --normalize -f /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/dodeca_bottom_cox.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/dodeca_top_cox.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_cox.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_plus.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_t.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_x.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_cox.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_cox_wide.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_plus.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_x.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_deadcat.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_h.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_plus.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_s250aq.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_vtail.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_wide.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x_cw.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x_pusher.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_y.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/tri_y.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/twin_engine.toml -o mixer_multirotor_normalized.generated.h
Failed to import toml: No module named 'toml'

You may need to install it using:
    pip3 install --user toml

[4/796] Generating mixer_multirotor_6dof.generated.h
FAILED: src/lib/mixer/MultirotorMixer/mixer_multirotor_6dof.generated.h 
cd /home/kannachan/px4/PX4-Autopilot/build/px4_sitl_default/src/lib/mixer/MultirotorMixer && /usr/bin/python3 /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/tools/px_generate_mixers.py --sixdof -f /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/dodeca_bottom_cox.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/dodeca_top_cox.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_cox.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_plus.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_t.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/hex_x.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_cox.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_cox_wide.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_plus.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/octa_x.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_deadcat.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_h.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_plus.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_s250aq.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_vtail.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_wide.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x_cw.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_x_pusher.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/quad_y.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/tri_y.toml /home/kannachan/px4/PX4-Autopilot/src/lib/mixer/MultirotorMixer/geometries/twin_engine.toml -o mixer_multirotor_6dof.generated.h
Failed to import toml: No module named 'toml'

You may need to install it using:
    pip3 install --user toml

[5/796] Generating uORB topic headers
FAILED: uORB/topics/uORBTopics.hpp uORB/topics/actuator_armed.h uORB/topics/actuator_controls.h uORB/topics/actuator_outputs.h uORB/topics/adc_report.h uORB/topics/airspeed.h uORB/topics/airspeed_validated.h uORB/topics/airspeed_wind.h uORB/topics/battery_status.h uORB/topics/camera_capture.h uORB/topics/camera_status.h uORB/topics/camera_trigger.h uORB/topics/cellular_status.h uORB/topics/collision_constraints.h uORB/topics/collision_report.h uORB/topics/commander_state.h uORB/topics/control_allocator_status.h uORB/topics/cpuload.h uORB/topics/debug_array.h uORB/topics/debug_key_value.h uORB/topics/debug_value.h uORB/topics/debug_vect.h uORB/topics/differential_pressure.h uORB/topics/distance_sensor.h uORB/topics/ekf2_timestamps.h uORB/topics/ekf_gps_drift.h uORB/topics/esc_report.h uORB/topics/esc_status.h uORB/topics/estimator_baro_bias.h uORB/topics/estimator_event_flags.h uORB/topics/estimator_innovations.h uORB/topics/estimator_optical_flow_vel.h uORB/topics/estimator_selector_status.h uORB/topics/estima [... cut ...]
I'm working in an anaconda environment and I had tried to install jiinja2, toml, and empy but it said it was already installed and met all the requirements. Then, I tried to uninstall all 3 packages and reinstall by typing pip3 install --user jinga2 toml empy. I run sudo make px4_sitl jmavsim again and the same error messages appeared. I uninstall the packages again and try to install it by typing pip install jinja2 toml empy which worked on my other machine but for some reason I still get the same error messages

dagar, 2021-08-17:
Can you try `make px4_sitl_default install_python_requirements`?

It installs the python requirements within the build with whichever python environment it found.

HatedFate, 2021-08-17:
> Can you try `make px4_sitl_default install_python_requirements`?
> 
> It installs the python requirements within the build with whichever python environment it found.

[0/1] cd /home/kannachan/px4/PX4-Autopilot/build/px4_sitl_default...nt /home/kannachan/px4/PX4-Autopilot/Tools/setup/requirements.txt
/usr/bin/python3: No module named pip
FAILED: CMakeFiles/install_python_requirements 
cd /home/kannachan/px4/PX4-Autopilot/build/px4_sitl_default && /usr/bin/python3 -m pip install --requirement /home/kannachan/px4/PX4-Autopilot/Tools/setup/requirements.txt
ninja: build stopped: subcommand failed.
Makefile:225: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1

This is the error i got from trying to run the command:
sudo make px4_sitl_default install_python_requirements

closed 2021-08-17 by dagar

LuoRongLuoRong, 2022-08-01:
> Can you try `make px4_sitl_default install_python_requirements`?
> 
> It installs the python requirements within the build with whichever python environment it found.

It works for me! Thanks~