PX4/PX4-Autopilot #18095: QGC:Parameters: abnormal display

gitfishup, 2021-08-18:
**Describe the bug**
[image]

The QGC version 
[image]

The version of Firmware of CUAVv5 flight controller:
commit :ff39e27e2d093001e777f48e2e178368c90a9ae5
[image]

From this commit , QGC:Parameters: abnormal display

dagar, 2021-08-18:
Thanks for reporting, I'll try to reproduce.

dagar, 2021-08-19:
Should be fixed in https://github.com/PX4/PX4-Autopilot/pull/18105.

closed 2021-08-19 by dagar